nanopb/nanopb #271: Can't import proto file using relative path using the CMake `nanopb_generate_cpp`

idelsink, 2017-07-19:
I'm using the CMake "nanopb_generate_cpp" macro but I can't import a `.proto` 
file from a relative directory in the `.proto` source.

See my example:

## Example

Input files:
```text
a.proto
subDir/b.proto
```

Contents of `a.proto`:
```proto
import "subDir/b.proto";
```

The generated output would be:
```text
/build/<path to output>/a.pb
/build/<path to output>/a.pb.c
/build/<path to output>/a.pb.h
/build/<path to output>/b.pb
/build/<path to output>/b.pb.c
/build/<path to output>/b.pb.h
```

But the `a.pb.h` file has an `#include "subDir/b.pb.h"`. 
This fails to compile because the header can't be found because there is no
`subDir` sub directory.

## Expected behavior

The expected behavior would be that the generated output is as follows:
```text
/build/<path to output>/a.pb
/build/<path to output>/a.pb.c
/build/<path to output>/a.pb.h
/build/<path to output>/subDir/b.pb
/build/<path to output>/subDir/b.pb.c
/build/<path to output>/subDir/b.pb.h
```

This would be the same behavior as protoc which has the `--proto_path` parameter 
to pass the root path of all the proto files.

## Proposed solutions

My proposed solution is to add an extra parameter to the `nanopb_generate_cpp`
to set a root path or use the cmake `CMAKE_CURRENT_LIST_DIR` variable.
This is used to keep the same directory structure in the generated output directory.
The user now needs to make sure to include all of those sub-directory's in the 
`target_include_directories` CMake command.
A command could be used to extract all the directory's from the output source/header files
and add them to a variable using something like the following (taken from [this](https://cmake.org/pipermail/cmake/2012-June/050674.html)):
```CMake
MACRO(HEADER_DIRECTORIES return_list)
    FILE(GLOB_RECURSE new_list *.h)
    SET(dir_list "")
    FOREACH(file_path ${new_list})
        GET_FILENAME_COMPONENT(dir_path ${file_path} PATH)
        SET(dir_list ${dir_list} ${dir_path})
    ENDFOREACH()
    LIST(REMOVE_DUPLICATES dir_list)
    SET(${return_list} ${dir_list})
ENDMACRO()
```
If you're open for this change I can submit a PR.

PetteriAimonen, 2017-07-19:
I'm quite ignorant on cmake front. What you say seems sane and I'm willing to accept a pull request.

@kylemanna, @kfitch : Any thoughts?

idelsink, 2017-07-20:
If I'm gonna edit this would it be ok too also edit the [cmake_simple](https://github.com/nanopb/nanopb/tree/master/examples/cmake_simple) example to have a subdirectory example? 

This would than have an extra proto file in a subdirectory to get the following:
```proto
syntax = "proto2";
import "unlucky/unlucky.proto";

message SimpleMessage {
    required int32 lucky_number = 1;
    required UnluckyNumber unlucky_number = 2;
}
```

Or would it be better to add an extra test to the `tests` directory that does the above as a test case?

PetteriAimonen, 2017-07-20:
Hmm, currently there isn't much stuff testing the different build systems. But that might not be very easy to do through scons. I guess modifying cmake_simple works for now.

kfitch, 2017-07-22:
@idlesink I have made an attempt at adding an optional "RELPATH" parameter to NANOPB_GENERATE_CPP in my nanopb branch https://github.com/kfitch/nanopb

The intent is you would do something like
NANOPB_GENERATE_CPP( SRCS HDRS proto/a.proto proto/sub/b.proto RELPATH proto)

Could you give it a try and give some feedback?

idelsink, 2017-07-22:
After trying it out in my project I was happy to see that it was working as expected.

What you could do, and I recommend this, is to add this feature to the tests and/or an example and make sure to update the documentation at the top of the "FindNanopb.cmake" file.

kfitch, 2017-07-22:
I added some docs and an example in my branch. I will go ahead and submit a pull request.

idelsink, 2017-07-24:
After updating my own project with the new merge and testing it again, everything seems to be working fine.
I think that we can close this issue for it has been fixed.

closed 2017-07-24 by idelsink