nanopb/nanopb #274: Deterministic output / option to suppress timestamp

ezyang, 2017-08-22:
By default, nanopb generates a header with the following contents:

```
/* Generated by nanopb-0.3.9-dev at Tue Aug 22 13:10:44 2017. */
```

This is not good for reproducibility, since it means that every invocation of nanopb will generate different contents. It would be great if there were a way to suppress the timestamp here (or, if I may advocate a bit forcefully, to NOT include it at all.)

PetteriAimonen, 2017-08-23:
Hmm yeah, might make sense to remove the timestamp completely.

But there is already an option, -T to nanopb_generator.py or `protoc --nanopb_out=-T:.`

ezyang, 2017-08-23:
Thanks! It is too bad `-T` doesn't show up when you run `protoc --help` with the nanopb plugin

PetteriAimonen, 2017-08-23:
Yeah, protoc doesn't let plugins to provide their own help text.

closed 2017-08-23 by PetteriAimonen