nanopb/nanopb #276: What is the deal with the tag numbers?

Larswad, 2017-08-23:
Maybe this is an implementation detail or such, but from the .proto file point of view, I can't really see the point of specifying a tag number for each field. I mean, why would anyone do something different than increasing each subsequent member by 1 for all field?
I can see why each field need an ID, but those numbers could have been generated automatically anyway, or? Is it for handling versioning, so that an extended file doesn't change original numbers in case new fields are added in the middle of the struct, and then get new field numbers?

Sorry if this is a stupid question, but I just don't get it. I have only seen the field number mentioned in the "limitations" section of the documentation. But not really an explanation why they have to be specified.

PetteriAimonen, 2017-08-23:
It is for versioning, like in all protobuf implementations.

Maybe this document is helpful: https://developers.google.com/protocol-buffers/docs/proto

Larswad, 2017-08-24:
Thanks! I started to suspect it had to do with that.
Are the tag numbers in proto buffers always mandatory to manually specify, or can protoc autogenerate them? Can't see that in that document.
See, in our project we will make sure the structure versions are always compatible between both communicating sides, so we won't need that tag number uniqueness (they can be autogenerated upon compile time). Maybe that could be some useful option to add during generation (but by default off, of course)?

PetteriAimonen, 2017-08-24:
Well, I cannot add it on nanopb side because Google's protoc handles parsing the file.

And honestly, I don't think it is a particularly good idea - filling in the tag numbers is not much trouble and it allows much better version compatibility - something that may not be important to you currently, but might be in the future.

Larswad, 2017-08-24:
Yes, it's not a deal breaker or anything on my side, it was simply a thought to reduce the amount of unneeded text and maintenance. . And I wasn't sure to what extent protoc did the whole parsing. But I understand now this numbering is inherent to the whole protobuf concept. 

I am happy with all this, thanks. BTW, I hope the nanopb streams  are binary compatible with regular protocol buffers (as long as not using anything outside of the protobuf specification)? 

Maybe the message id generation would fail the compilation if using  the regular protoc compiler. Because I like that feature very much.

PetteriAimonen, 2017-08-24:
All nanopb features should be compatible with other protobuf implementations.

closed 2017-09-15 by PetteriAimonen