nanopb/nanopb #277: pb_decode doesn't properly return failure on malformed input

vlovich, 2017-09-02:
echo 9fea | xxd -r -p | protoc --decode_raw
Failed to parse input.

C++ protobuf rejects this input too.

pb_decode accepts this input because of a bug in pb_decode_tag which fails to decode the tag but because it consumed the whole input so it sets eof = true which successfully exits the decode loop.

Seems hard to exploit as a security issue but it might be a good idea to not return success on malformed input.

vlovich, 2017-09-02:
One fix might be to have the integer decode routines unwind the stream before returning failure.  Alternatively, pb_decode_tag could return with eof set to true before it decodes the tag & leave eof = false if it failed to decode the tag.

PetteriAimonen, 2017-09-02:
Hmm, there are also other cases where nanopb doesn't detect malformed input while C++ protobuf does, such as mismatched field type vs. wire type.

In general I don't think these are worth fixing, but this specific one is probably quite simple to fix so might be worth it.

vlovich, 2017-09-02:
This was my fix:
    
         if (!pb_decode_varint32(stream, &temp))
         {
    -        if (stream->bytes_left == 0)
    -            *eof = true;
    -
             return false;
         }

Seems safe.  This function never gets called without bytes remaining in the stream.  Failing to decode the varint means the stream terminated prematurely.

PetteriAimonen, 2017-09-15:
Thanks, this should be fixed now.

Your proposed patch didn't quite work, as bytes_left is only accurate for memory buffers. It fails e.g. the `tests/basic_stream` test case. But moving the `eof` check into `pb_decode_varint32_eof()` works to allow `eof` only to occur on the first byte of the varint.

PetteriAimonen, 2017-09-23:
Fix released in nanopb-0.3.9.

closed 2017-09-23 by PetteriAimonen