nanopb/nanopb #279: pb_decode can't decode a tag with large numbers

vlovich, 2017-09-02:
Input value: 88d4c394a303862edd239fd0ec1c

> cat test.proto
syntax = "proto2";
message Test {
  optional uint32 f = 106044737;
  optional fixed32 g = 571;
}

> echo 88d4c394a303862edd239fd0ec1c | xxd -r -p | protoc --decode Test test.proto
g: 485281951
f: 5894

Decoding fails with nanopb because the varint32 decoder thinks there's been an overflow.

PetteriAimonen, 2017-09-02:
To me, this seems more like an invalid protobuf message which protoc just doesn't detect.

The number encoded in `88 d4 c3 94 a3 03` is `0x1a3290ea08` which gives a tag number of 14064688449, which is more than 536,870,911, which is the maximum [in the specification](https://developers.google.com/protocol-buffers/docs/proto#assigning-tags). Google's implementation silently truncates this to 32 bits, while nanopb issues an error. Either behaviour seems fine to me, as protobuf is not really designed to reliably detect corrupted messages - one would use e.g. checksum for that purpose.

Also, if possible, please try to slow down your issue reporting pace. I'm pretty short on time that I can spend on nanopb nowadays, but I do not like to build a big queue of unanswered issue reports either.

vlovich, 2017-09-02:
My bad.  Thanks for the quick reply.

closed 2017-09-02 by vlovich