nanopb/nanopb #280: proto3 and empty fixed size arrays

MrCircuit, 2017-09-06:
I've got a message that contains several sub messages. An example of one of these sub messages:

```
// The structure of the IP configuration message type.
message IPConfigType
{
    bytes APN = 1 [(nanopb).max_size = 16, (nanopb).fixed_length = true];
    bytes IP = 2 [(nanopb).max_size = 4, (nanopb).fixed_length = true];
    uint32 ServerPort = 3; 
    uint32 Protocol = 4;
}
```

The encoder does not skip these sub messages, if they are left empty (init default) but contain fixed size arrays like the one above. I've tracked it down to the code of pb_encode.c, line 249. This default check will always result in false as data_size is always greater zero thanks to fixed size arrays.

Should we alternatively check bytewise if each of them is zero or is there another way to detect empty arrays?

PetteriAimonen, 2017-09-06:
Hmm, a fixed size array will always encode as N bytes. But if we left out the field altogether, it would decode as 0 bytes, not N zero bytes on the other end. So I think fixed size arrays must always be encoded, even if all bytes are 0.

One way to to skip it to specify `(nanopb).proto3 = false` on the field and then set the `has_field` to true/false.

MrCircuit, 2017-09-06:
What is the difference between the method `pb_check_proto3_default_value` that is used to find out if the field is implicitly skipped and the method proposed by you using `(nanopb).proto3 = false` ? For me, as a protobuf rookie, they seem to lead to the same result in the data stream.

PetteriAimonen, 2017-09-06:
`pb_check_proto3_default_value()` checks the default value according to https://developers.google.com/protocol-buffers/docs/proto3#default . So for bytes, default value is empty, and fixed length array is not empty, so it cannot be skipped.

If you specify `(nanopb).proto3 = false`, you get two fields in the generated struct: "pb_byte_t IP[4];" and "bool has_IP;". You then have to set the "has_IP" to true if you want to encode the field, or to false if you don't want to encode it. It will lead to the same effect in the encoded message as if `pb_check_proto3_default_value()` returned true for the zero-bytes case, but due to how the spec works, it cannot be changed to do that.

MrCircuit, 2017-09-06:
Ok. Thank you for the explanation.

closed 2017-09-06 by MrCircuit