nanopb/nanopb #281: Error when defining fixed_length

iamthad, 2017-09-13:
When I define `fixed_length` for a field in a `.options` file, e.g.

```
MESSAGE.field fixed_length:1
```

I get the following error:
```
Traceback (most recent call last):
  File "~/nanopb/generator/nanopb_generator.py", line 1664, in <module>
    main_plugin()
  File "~/nanopb/generator/nanopb_generator.py", line 1644, in main_plugin
    other_files[fdesc.name] = parse_file(fdesc.name, fdesc, options)
  File "~/nanopb/generator/nanopb_generator.py", line 1519, in parse_file
    f = ProtoFile(fdesc, file_options)
  File "~/nanopb/generator/nanopb_generator.py", line 1058, in __init__
    self.parse()
  File "~/nanopb/generator/nanopb_generator.py", line 1084, in parse
    self.messages.append(Message(names, message, message_options))
  File "~/nanopb/generator/nanopb_generator.py", line 865, in __init__
    field = Field(self.name, f, field_options)
  File "~/nanopb/generator/nanopb_generator.py", line 370, in __init__
    self.enc_size = varint_max_size(self.max_size) + self.max_size
  File "~/nanopb/generator/nanopb_generator.py", line 127, in varint_max_size
    max_value = 2**64 - max_value
TypeError: unsupported operand type(s) for -: 'long' and 'NoneType'
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```

PetteriAimonen, 2017-09-14:
Hmm, maybe you have not defined the separate `max_size` option that is required to be used with `fixed_length`?

But the error message should be improved.

iamthad, 2017-09-14:
Indeed, I did not define max_size. I must have missed where that requirement is documented.

PetteriAimonen, 2017-09-23:
Fix released in nanopb-0.3.9.

closed 2017-09-23 by PetteriAimonen