nanopb/nanopb #289: Support for Map data type

winterscar, 2017-10-01:
Hi, is there any (planned) support for the map data type?

PetteriAimonen, 2017-10-01:
Currently there is no special support for it. Main issue is that C doesn't have any widely used standard format for maps/dictionaries, so it is not quite clear how nanopb should implement it.

However the `Map` datatype in protobuf is itself just a repeated submessage, so it can be used as described here in `Backwards compatibility`:
https://developers.google.com/protocol-buffers/docs/proto3#maps

winterscar, 2017-10-01:
@PetteriAimonen , thanks for taking the time to reply :) I've seen that section, but I'm having a hard time implementing in code. Do you possibly have any example or psuedocode I could take a look at?

PetteriAimonen, 2017-10-01:
There is now:
https://github.com/nanopb/nanopb/tree/master/tests/map

winterscar, 2017-10-01:
Dude, you are a *legend*. Thank you so much :)

closed 2017-10-01 by winterscar

Hacker-CB, 2021-11-03:
Hello,

Is there any example to encode/decode maps with callback?

For example, if I have next map:

`map<string, string> myMap`

Is the next encoder correct?

```c++
std::string key = "mykey";
std::string value = "myvalue";

if (!pb_encode_tag_for_field(stream, field))
    return false;

if (!pb_encode_string(stream, (pb_byte_t *)(key.data()), key.size()))
    return false;

if (!pb_encode_string(stream, (pb_byte_t *)(value.data()), value.size()))
    return false;
```

PetteriAimonen, 2021-11-03:
@Hacker-CB No, that is not exactly right. The map internally has this format:

<pre>
message myMapEntry {
     string key = 1;
     string value = 2;
}
message ... {
     repeated myMapEntry myMap = ..;
}
</pre>

 There are two ways you could go about it:

1. Assign maximum sizes to the key and value and use `pb_encode_submessage()` to encode it as a whole.
2. Compute the size of the submessage yourself and call `pb_encode_varint()` to encode the length and for each of the key and value fields call `pb_encode_tag` and `pb_encode_string`.

Hacker-CB, 2021-11-03:
@PetteriAimonen, thank you for the fast response.

So, my message looks like this:

```proto
message JHP_ErrorMessage {
    int32 errorCode = 1;
    map<string, string> errorArgs = 2; // Error arguments
}
```

Is this correct encoder?

```c++
using StringMap = std::map<std::string, std::string>;

StringMap myMap;

auto errorArgsEncoder = +[](pb_ostream_t *stream, const pb_field_t *field, void *const *arg) -> bool {
    StringMap *myMap = static_cast<StringMap *>(*arg);

    if (!pb_encode_varint(stream, myMap.size()))
        return false;

    for (auto& errorArg : *myMap){
        auto& key = errorArg.first;
        auto& value = errorArg.second;

        if (!pb_encode_tag_for_field(stream, field))
            return false;

        if (!pb_encode_string(stream, (pb_byte_t *)(key.data()), key.size()))
            return false;

        if (!pb_encode_tag_for_field(stream, field))
            return false;

        if (!pb_encode_string(stream, (pb_byte_t *)(value.data()), value.size()))
            return false;
    }
    return true;
};

JHP_ErrorMessage msg = {
        .errorArgs =  {.funcs={.encode=errorArgsEncoder}, .arg=(void*)&myMap}
};

pb_encode(stream, &JHP_ErrorMessage_msg, &msg)
```

And It is not clear for me how to make decoder, seems it will be much more compilcated.
Could you point me to the right way please? 

```c++
using StringMap = std::map<std::string, std::string>;

StringMap errorArgs;

auto errorArgsDecoder = +[](pb_istream_t *stream, const pb_field_t *field, void **arg) -> bool {
    StringMap *errorArgs = static_cast<StringMap *>(*arg);
    uint64_t mapSize;

    if (!pb_decode_varint(stream, &mapSize))
        return false;
    
    // what to do here?

    // next code I used to decode dynamic length string before, but it reads entire stream...
    
    //std::string str;
    //str.resize(stream->bytes_left);
    //if (!pb_read(stream, (uint8_t *) str.data(), stream->bytes_left)) {
    //    return false;
    //}

};

JHP_ErrorMessage msg = {
        .errorArgs = {.funcs={.decode=errorArgsDecoder}, .arg=(void *) &errorArgs}
};

pb_decode(stream, &JHP_ErrorMessage_msg, &msg);
```

PetteriAimonen, 2021-11-03:
@Hacker-CB Unfortunately I don't currently have enough time to make a more complete example of that. https://developers.google.com/protocol-buffers/docs/encoding is a useful read. For the encoder side, you can compare what it generates against the data generated by e.g. `protoc --encode`. `tests/raw_decode` can be useful for that.

Decoding is not much different, but you'll need to call `pb_decode_varint` to read the string sizes.

Hacker-CB, 2021-11-03:
@PetteriAimonen,

I found `JHP_ErrorMessage_ErrorArgsEntry` in the generated code. Sorry for the my stupidity :)

Encoding/decoding works fine now like for any other repeated message. 

I just used generated `JHP_ErrorMessage_ErrorArgsEntry` for `pb_encode_submessage()` and inner `pb_decode()` calls.

Thank you for the help.

**Final working encoder:**

```c++
PB_CB_ENCODER(errorArgsEncoder,ErrorMessage, self){
    for (auto &errorArg: self->errorArgs) {
        auto &key = errorArg.first;
        auto &value = errorArg.second;

        JHP_ErrorMessage_ErrorArgsEntry entry = {
                .key = PB_STRING_ENCODER(&key),
                .value = PB_STRING_ENCODER(&value)
        };

        if (!pb_encode_tag_for_field(stream, field))
            return false;

        if (!pb_encode_submessage(stream, &JHP_ErrorMessage_ErrorArgsEntry_msg, &entry))
            return false;
    }
    return true;
}};

JHP_ErrorMessage msg = {
        .errorCode = errorCode,
        .errorArgs = PB_ENCODE_CB(errorArgsEncoder,this)
};
```

**Final working decoder:**

```c++
using StringMap = std::map<std::string, std::string>;

StringMap errorArgs;

PB_CB_DECODER(errorArgsDecoder, StringMap, errorArgs){
    std::string key, value;
    JHP_ErrorMessage_ErrorArgsEntry entry = {
            .key = PB_STRING_DECODER(&key),
            .value = PB_STRING_DECODER(&value)
    };
    if (!pb_decode(stream, &JHP_ErrorMessage_ErrorArgsEntry_msg, &entry)) {
        return false;
    }
    (*errorArgs)[key] = value;

    return true;
}};

JHP_ErrorMessage msg = {
        .errorCode = 0,
        .errorArgs = PB_DECODE_CB(errorArgsDecoder,&errorArgs)
}; 
```   

<details>
  <summary>Common macros and helpers</summary>

```c++
#define PB_ENCODE_CB(cb,ctx){ .funcs = { .encode = cb }, .arg = (void*)ctx }
#define PB_DECODE_CB(cb,ctx){ .funcs = { .decode = cb }, .arg = (void*)ctx }

#define PB_STRING_ENCODER(strPtr) PB_ENCODE_CB(JHL::Proto::stringEncoderCallback,strPtr)
#define PB_STRING_DECODER(strPtr) PB_DECODE_CB(JHL::Proto::stringDecoderCallback,strPtr)

#define PB_CB_ENCODER(encoderName, argType, argName) \
    auto encoderName = +[](pb_ostream_t *stream, const pb_field_t *field, void * const *arg) -> bool { \
        argType* argName = static_cast<argType *>(*arg);

#define PB_CB_DECODER(decoderName, argType, argName) \
    auto decoderName = +[](pb_istream_t *stream, const pb_field_t *field, void **arg) -> bool { \
        argType* argName = static_cast<argType *>(*arg);

bool JHL::Proto::stringDecoderCallback(pb_istream_t *stream, const pb_field_t *field, void **arg) {
    std::string *str = static_cast<std::string *>(*arg);
    size_t len = stream->bytes_left;
    str->resize(len);
    if (!pb_read(stream, (uint8_t *) str->data(), len)) {
        return false;
    }
    return true;
}

bool JHL::Proto::stringEncoderCallback(pb_ostream_t *stream, const pb_field_t *field, void *const *arg) {
    const std::string *str = static_cast<const std::string *>(*arg);
    if (!pb_encode_tag_for_field(stream, field))
        return false;
    return pb_encode_string(stream, (const pb_byte_t *) str->c_str(), str->size());
}
```

</details>

PetteriAimonen, 2021-11-04:
@Hacker-CB Yeah, `protoc` generates that message transparently. Great to hear you got it working :)