nanopb/nanopb #291: When I try to define uint8_t as bool, nano-pb seems to have some problem.

tutengkwwl, 2017-10-06:
Hi all:

I use nano-pb on my MCU, and my complier dose't support 'stdbool'. So I have to define uint8_t as bool type.
When I define uint8_t as bool (To reduce RAM usage), the encoding data by MCU is not equal as my computer generated, and the data can't decode by MCU or computer yet.
But when I define uint32_t as bool, the nano-pb works well.

I don't know why this happens……
Is the bool must define as an uint32_t ?

PetteriAimonen, 2017-10-06:
Are you using C++ or only C?

tutengkwwl, 2017-10-07:
@PetteriAimonen Only C

PetteriAimonen, 2017-10-07:
It should work fine, as long as you do the typedef only in one place and remember to recompile all files, especially .pb.c files.

Also if you can post the un-decodable data and associated .proto file, it may offer some clues.

tutengkwwl, 2017-10-14:
Hi @PetteriAimonen .
 It's my fault. 
My  MCU‘s C complier has no STD head files, and I have't replace PB_SYSTEM_HEADER as "pb_syshdr.h". 
I'm sorry I have no time to find the real reason, but I suppose there maybe some issues about "offsetof" on my C complier.

Thanks for your great work! Now I can easyly use protobuf on my MCU.

closed 2017-10-14 by tutengkwwl