nanopb/nanopb #292: Apple Mach-O Linker Error

ahspadafora, 2017-10-20:
Running on iPhone 8 Plus simulator 
I am getting the following Apple Mach-O Linker Error in nanopb 

```
duplicate symbol _pb_field_iter_next in:
/Users/mac/Library/Developer/Xcode/DerivedData/waveR-fsbvdaitkxkofdefvyejiewlyefx/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/nanopb.build/Objects-normal/x86_64/pb_common 2.o
/Users/mac/Library/Developer/Xcode/DerivedData/waveR-fsbvdaitkxkofdefvyejiewlyefx/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/nanopb.build/Objects-normal/x86_64/pb_common.o
duplicate symbol _pb_field_iter_begin in:
/Users/mac/Library/Developer/Xcode/DerivedData/waveR-fsbvdaitkxkofdefvyejiewlyefx/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/nanopb.build/Objects-normal/x86_64/pb_common 2.o
/Users/mac/Library/Developer/Xcode/DerivedData/waveR-fsbvdaitkxkofdefvyejiewlyefx/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/nanopb.build/Objects-normal/x86_64/pb_common.o
duplicate symbol _pb_field_iter_find in:
/Users/mac/Library/Developer/Xcode/DerivedData/waveR-fsbvdaitkxkofdefvyejiewlyefx/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/nanopb.build/Objects-normal/x86_64/pb_common 2.o
/Users/mac/Library/Developer/Xcode/DerivedData/waveR-fsbvdaitkxkofdefvyejiewlyefx/Build/Intermediates.noindex/Pods.build/Debug-iphonesimulator/nanopb.build/Objects-normal/x86_64/pb_common.o
ld: 3 duplicate symbols for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

PetteriAimonen, 2017-10-20:
Hmm, it seems you have two copies of nanopb there, one called "pb_common 2.o" and another called "pb_common.o". Do you know where that "pb_common 2.o" might be coming from?

ahspadafora, 2017-10-23:
I resolved the issue by deleting my xcworkspace and removing all podfiles, I'm not sure what caused the duplication I must have manually added files that also had nanopb

closed 2017-10-23 by ahspadafora