nanopb/nanopb #295: default value for enum in proto2 not correct

dvescovi1, 2017-10-31:
In the protobuf 2 language spec reads the following:
" For enums, the default value is the first value listed in the enum's type definition."
if I do something like:
enum MyPayload {
  test1 = 5;
  test2 = 7;
}

message MyTest {
  required MyPayload payload = 1;
}

the resulting initializer end up looking like:
#define MyTest_init_default                      {(MyPayload)0}

which is incorrect

The workaround would be to explicitly define a default like:

message MyTest {
  required MyPayload payload = 1 [default = test2];
}
which does seem to generate the init correctly.

I realize in protobuf3 language spec this statement is also present but they have an additional requirement that all enums must start at zero. in which case it would be ok except you would also have to remove the "required" from the message definition.
But never the less the protobuf2 init for enums is not correct.

PetteriAimonen, 2017-11-01:
Indeed, fortunately it should be quite simple fix in the generator.

dvescovi1, 2017-11-01:
looking forward to a fix

PetteriAimonen, 2017-11-01:
It will probably be a while until I'll have time to get to it.

dvescovi1, 2017-12-09:
I was using this as my test file (defaultTest.proto):
syntax = "proto2";

enum MyPayload {
 test1 = 5;
 test2 = 7;
 }

message MyTest1 {
 required MyPayload payload = 1;
 }

message MyTest2 {
 required MyPayload payload = 1 [default = test2];
 }

I looked into this and I see if I change the following in the get_initializer function at approx. line 452:
from:
           elif self.pbtype in ('ENUM', 'UENUM'):
                inner_init = '(%s)0' % self.ctype
 
to:
            elif self.pbtype in ('ENUM', 'UENUM'):
                inner_init = '_%s_MIN' % self.ctype

This will result in the correct xxx_init_default and xxx_init_zero strings in the resulting defaultTest.pb.h file:
/* Enum definitions */
typedef enum _MyPayload {
    MyPayload_test1 = 5,
    MyPayload_test2 = 7
} MyPayload;
#define _MyPayload_MIN MyPayload_test1
#define _MyPayload_MAX MyPayload_test2
#define _MyPayload_ARRAYSIZE ((MyPayload)(MyPayload_test2+1))
...

/* Initializer values for message structs */
#define MyTest1_init_default                     {_MyPayload_MIN}
#define MyTest2_init_default                     {MyPayload_test2}
#define MyTest1_init_zero                        {_MyPayload_MIN}
#define MyTest2_init_zero                        {_MyPayload_MIN}

My guess this should be ok for proto2 and proto3 as in the proto3 case it will also resolve to zero if enum start at zero.

Does this look like a valid fix?

PetteriAimonen, 2017-12-09:
Hm, not sure what should happen with _zero initializer for an enum that does not have zero value. But I guess the fix makes sense.

PetteriAimonen, 2017-12-09:
@dvescovi1 Committed, thanks :)

dvescovi1, 2017-12-09:
Thanks for the check.
My guess the "xxx_init_zero" name is now slightly misleading as it seems "xxx_init" might make it more understandable.

In my test case I would use it like:

MyTest1 msg = MyTest1_init_default;
or
MyTest1 msg = MyTest1_init;    // instead of ... = MyTest1_init_zero

...minor detail, I can live with ether ;)