nanopb/nanopb #296: Install path of Python plugin incorrect when using relative CMAKE_INSTALL_PREFIX

dflogeras, 2017-11-08:
Hi there, I've noticed that if I do an out-of-tree build with CMake, and then configure CMAKE_INSTALL_PREFIX to be "../", the nanopb library and include files go to the correct location, but the python plugins end up in "../../".  Similarly, if I specify CMAKE_INSTALL_PREFIX="../../", the Python plugins end up in "../../../../" (as long as I have write permission.

To reproduce, start with a clean 0.3.9 tree.
cd nanopb
mkdir build
cmake ..
ccmake .. (edit the CMAKE_INSTALL_PATH to be "..")
[configure, generate]
make
make install

Note it seems to work, if you configure with "cmake .. -DCMAKE_PREFIX_PATH=..", since it turns it into an absolute path first.

PetteriAimonen, 2017-11-08:
Would you like to make a patch and pull request to fix this? I don't know much about CMake myself.

dflogeras, 2017-11-09:
I can sure try to, I'll have a look in the next few days. Thanks for nanopb!

commit 298b6b9f (2017-11-27): Merge pull request #298 from dflogeras/cmake_python_instdir_fix

Fix #296 - install path of Python plugins with CMake.
commit cae980ca (2017-11-27): Fix #296 - install path of Python plugins with CMake.

No need of specifying the CMAKE_PREFIX_PATH to python when asking for
sysconfig.get_python_lib since everything is installed relative to
CMAKE_PREFIX_PATH anyway.
closed 2017-11-27 by PetteriAimonen