nanopb/nanopb #297: TypeError: __init__() got an unexpected keyword argument 'file' on OS X

ydesgagn, 2017-11-26:
Hi,

I'm trying to use nanopb to generate code with protobuf version 3. I always get the following error:

`
         ****************************************************************************
         *** Got TypeError when importing the protocol definitions for generator. ***
         *** This usually means that the protoc in your path doesn't match the    ***
         *** Python protobuf library version.                                     ***
         ***                                                                      ***
         *** Please check the output of the following commands:                   ***
         *** which protoc                                                         ***
         *** protoc --version                                                     ***
         *** python -c 'import google.protobuf; print(google.protobuf.__file__)'  ***
         ****************************************************************************
    
Traceback (most recent call last):
  File "/usr/local/Cellar/nanopb-generator/0.3.9/libexec/nanopb_generator.py", line 33, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "/usr/local/Cellar/nanopb-generator/0.3.9/libexec/proto/nanopb_pb2.py", line 121, in <module>
    options=None, file=DESCRIPTOR)
TypeError: __init__() got an unexpected keyword argument 'file'
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
`

The environment is the following:
`
$ protoc --version 
libprotoc 3.5.0

$ python -c 'import google.protobuf; print(google.protobuf.__version__)'
3.5.0.post1

$ pip show protobuf
Name: protobuf
Version: 3.5.0.post1
Summary: Protocol Buffers
Home-page: https://developers.google.com/protocol-buffers/
Author: protobuf@googlegroups.com
Author-email: protobuf@googlegroups.com
License: 3-Clause BSD License
Location: /Library/Python/2.7/site-packages
Requires: setuptools, six

$ python -V
Python 2.7.10
`

I can generate without any issues for --c_out=. --cpp_out=. --go_out=. --ruby_out=. but not for --nanopb_out=.

Any help would be appreciated.

Thanks

ydesgagn, 2017-12-02:
You need to use the homebrew python on OS X to make it work. With the default OS X python, the libraries are not in the correct path that is hardcoded into the brew package.

closed 2017-12-02 by ydesgagn

PetteriAimonen, 2017-12-02:
Nice to hear that you got it figured out :)