nanopb/nanopb #300: STM32 IAR8.2 - Does not pass unit tests.

DJAGER47, 2017-12-14:
When checking encode_unittests.c fails with the following error in the console.

```
----Test pb_encode with int32 array----
�[31;1mFAILED:�[22;39m ..\..\..\FOTO_2load_v1.2\Src\main.c:127 WRITES(pb_encode(&s, IntegerArray_fields, &msg), "\x0A\x05\x01\x02\x03\x04\x05")
�[32;1mOK:�[22;39m WRITES(pb_encode(&s, IntegerArray_fields, &msg), "")
�[32;1mOK:�[22;39m !pb_encode(&s, IntegerArray_fields, &msg)

----Test pb_encode with float array----
�[31;1mFAILED:�[22;39m ..\..\..\FOTO_2load_v1.2\Src\main.c:144 WRITES(pb_encode(&s, FloatArray_fields, &msg), "\x0A\x04\x00\x00\xc6\x42")
�[32;1mOK:�[22;39m WRITES(pb_encode(&s, FloatArray_fields, &msg), "")
�[32;1mOK:�[22;39m !pb_encode(&s, FloatArray_fields, &msg)

----Test pb_encode with packed array in a submessage.----
�[31;1mFAILED:�[22;39m ..\..\..\FOTO_2load_v1.2\Src\main.c:160 WRITES(pb_encode(&s, IntegerContainer_fields, &msg), "\x0A\x07\x0A\x05\x01\x02\x03\x04\x05")

----Test pb_encode_delimited.----
�[31;1mFAILED:�[22;39m ..\..\..\FOTO_2load_v1.2\Src\main.c:170 WRITES(pb_encode_delimited(&s, IntegerContainer_fields, &msg), "\x09\x0A\x07\x0A\x05\x01\x02\x03\x04\x05")

----Test pb_get_encoded_size.----
�[31;1mFAILED:�[22;39m ..\..\..\FOTO_2load_v1.2\Src\main.c:179 pb_get_encoded_size(&size, IntegerContainer_fields, &msg) && size == 9

----Test string pointer encoding.----
�[32;1mOK:�[22;39m WRITES(pb_encode(&s, StringPointerContainer_fields, &msg), "\x0a\x00")
�[31;1mFAILED:�[22;39m ..\..\..\FOTO_2load_v1.2\Src\main.c:196 WRITES(pb_encode(&s, StringPointerContainer_fields, &msg), "\x0a\x01Z")

Some tests FAILED!
```
The rest of the tests are passed successfully.
What could be the reason?

PetteriAimonen, 2017-12-14:
Hmm, this seems like some trouble with the `pb_size_t` typedef. If you are using `PB_FIELD_32BIT` or `PB_FIELD_16BIT` compilation options, make sure that they apply to all files in your project.

Otherwise, could you set a breakpoint after the failed test cases and print out the data in `buffer` and the message length in `s`?

DJAGER47, 2017-12-15:
I don't use PB_FIELD_32BIT and PB_FIELD_16BIT.
I can write data buffer. Where to put the stop point?
```
                  uint8_t buffer[10];
	          pb_ostream_t s;
	          IntegerArray msg = {5, {1, 2, 3, 4, 5}};

	          COMMENT("Test pb_encode with int32 array")
	          TEST(WRITES(pb_encode(&s, IntegerArray_fields, &msg), "\x0A\x05\x01\x02\x03\x04\x05"))

	          msg.data_count = 0;
	          TEST(WRITES(pb_encode(&s, IntegerArray_fields, &msg), ""))

	          msg.data_count = 10;
	          TEST(!pb_encode(&s, IntegerArray_fields, &msg))
```

DJAGER47, 2017-12-15:
[Variables](http://radikal.ru/lfp/s014.radikal.ru/i326/1712/df/65aa8252340a.jpg/htm)
[Point](http://radikal.ru/lfp/s019.radikal.ru/i631/1712/b0/0c7f54942565.jpg/htm)

PetteriAimonen, 2017-12-15:
Hmm yeah, it still looks like some compilation option is different between unittestproto.pb.c and the main c file. Can you also print out the `IntegerArray_fields` variable?

DJAGER47, 2017-12-15:
[IntegerArray_fields](http://i062.radikal.ru/1712/88/dadbf4c13149.jpg)

PetteriAimonen, 2017-12-15:
Sorry for the trouble of remote debugging; I'll need a few more things:

`&msg`
`sizeof(msg.data_count)`
`&msg.data_count`
`&msg.data[0]`
`&msg.data[1]`

DJAGER47, 2017-12-15:
You need the addresses of the variables?

DJAGER47, 2017-12-15:
[addresses](http://s010.radikal.ru/i314/1712/6f/e0311442ba72.jpg)

DJAGER47, 2017-12-15:
Tested in simulator, not a real processor. The result is the same. :(

PetteriAimonen, 2017-12-15:
Thanks.

So in `IntegerArray_fields`, `data_offset` is 4, which means that address of `msg.data` should be 4 bytes from `msg`. But actually it appears to be 1.

Because `unittestsproto.pb.c` fills in `data_offset` using the `offsetof` macro, the only way this can happen is if the compilation options differ.

In this case, it seems that the struct packing differs. I suggest you recompile `unittestproto.pb.c`, making sure that you use the same compilation options for all files.

If that doesn't help, you can try debugging further by copying the `IntegerArray_fields` definition from .pb.c file to your main file with different name, and using that in the `pb_encode()` call instead. Then atleast the compilation options should always be the same if they are defined in the same file.

DJAGER47, 2017-12-15:
[download](https://yadi.sk/d/RH6zXd1l3Qeg4o)
Two projects for mingw(untitled) and iar.
The code in untitled, written in mainwindow.cpp
The code in iar CanParser.c

DJAGER47, 2017-12-15:
When you build with macro PB_FIELD_32BIT. Unit tests passed. But binary messages are different from harvested under minGW.

PetteriAimonen, 2017-12-15:
Unfortunately I don't think I'll have enough time to try to run that any time soon.

DJAGER47, 2017-12-20:
Gathered project under the GNU ARM Embedded Toolchain. The error is not disappeared. The data does not differ from the model in IAR. There is an assumption that the error of the processor HAL or libraries.

DJAGER47, 2017-12-21:
The problem was found. The second programmer, who wrote the analysis of our Protocol, used #pragma pack(push,1) and forgot to close it. For this reason, the following code worked correctly.

closed 2017-12-21 by DJAGER47

PetteriAimonen, 2017-12-21:
Nice to hear that you found the problem :)

DJAGER47, 2017-12-21:
Thanks for the help