nanopb/nanopb #301: How to init  pb_callback_t variable in struct?

aeris-ming, 2018-01-03:
Why the type I set in proto file become pb_callback_t in header file?
And How to operate it?
Thanks!
1.edit my prog.proto file
syntax = "proto3";

 package lib;

 message Trigger {
    string Time = 1;
    float Value = 2;
 }

 // Used by the device's function "program"
 message Program {
    enum Days {
        Monday = 0;
        Tuesday = 1;
        Wednesday = 2;
        Thursday = 3;
        Friday = 4;
        Saturday = 5;
        Sunday = 6;
    }

    repeated Days ActiveDays = 1;
    repeated Trigger Triggers = 2;
 }

2.run cmd ../generator-bin/protoc  --nanopb_out=./ ./prog.proto

3.get prog.pb.h file

#ifndef PB_LIB_PROG_PB_H_INCLUDED
#define PB_LIB_PROG_PB_H_INCLUDED
#include <pb.h>

/* @@protoc_insertion_point(includes) */
#if PB_PROTO_HEADER_VERSION != 30
#error Regenerate this file with the current version of nanopb generator.
#endif

#ifdef __cplusplus
extern "C" {
#endif

/* Enum definitions */
typedef enum _lib_Program_Days {
    lib_Program_Days_Monday = 0,
    lib_Program_Days_Tuesday = 1,
    lib_Program_Days_Wednesday = 2,
    lib_Program_Days_Thursday = 3,
    lib_Program_Days_Friday = 4,
    lib_Program_Days_Saturday = 5,
    lib_Program_Days_Sunday = 6
} lib_Program_Days;
#define _lib_Program_Days_MIN lib_Program_Days_Monday
#define _lib_Program_Days_MAX lib_Program_Days_Sunday
#define _lib_Program_Days_ARRAYSIZE ((lib_Program_Days)(lib_Program_Days_Sunday+1))

/* Struct definitions */
typedef struct _lib_Program {
    pb_callback_t ActiveDays;
    pb_callback_t Triggers;
/* @@protoc_insertion_point(struct:lib_Program) */
} lib_Program;

typedef struct _lib_Trigger {
    pb_callback_t Time;
    float Value;
/* @@protoc_insertion_point(struct:lib_Trigger) */
} lib_Trigger;

/* Default values for struct fields */

/* Initializer values for message structs */
#define lib_Trigger_init_default                 {{{NULL}, NULL}, 0}
#define lib_Program_init_default                 {{{NULL}, NULL}, {{NULL}, NULL}}
#define lib_Trigger_init_zero                    {{{NULL}, NULL}, 0}
#define lib_Program_init_zero                    {{{NULL}, NULL}, {{NULL}, NULL}}

/* Field tags (for use in manual encoding/decoding) */
#define lib_Program_ActiveDays_tag               1
#define lib_Program_Triggers_tag                 2
#define lib_Trigger_Time_tag                     1
#define lib_Trigger_Value_tag                    2

/* Struct field encoding specification for nanopb */
extern const pb_field_t lib_Trigger_fields[3];
extern const pb_field_t lib_Program_fields[3];

/* Maximum encoded size of messages (where known) */
/* lib_Trigger_size depends on runtime parameters */
/* lib_Program_size depends on runtime parameters */

/* Message IDs (where set with "msgid" option) */
#ifdef PB_MSGID

#define PROG_MESSAGES \

#endif

#ifdef __cplusplus
} /* extern "C" */
#endif

PetteriAimonen, 2018-01-03:
If you can determine a maximum count for the fields, it is best to use a field option e.g. `max_count: 10`, and then you'll get a simple array. Callbacks are only needed if you have an unlimited number of fields.

This might help some:
https://jpa.kapsi.fi/nanopb/docs/concepts.html#data-types

aeris-ming, 2018-01-04:
@PetteriAimonen Thanks versy much!

closed 2018-01-04 by aeris-ming