nanopb/nanopb #303: GCC 7.2 LTO warning

torfinnberset, 2018-01-04:
When compiling nanbopb with the gcc-arm-none-eabi 2017q4 toolchain (https://developer.arm.com/open-source/gnu-toolchain/gnu-rm) I get the following warning when linking with LTO enabled:

```
nanopb/pb_encode.h:164:6: warning: type of 'pb_encode_submessage' does not match original declaration [-Wlto-type-mismatch]
 bool pb_encode_submessage(pb_ostream_t *stream, const pb_field_t fields[], const void *src_struct);
      ^
nanopb/pb_encode.c:675:18: note: 'pb_encode_submessage' was previously declared here
 bool checkreturn pb_encode_submessage(pb_ostream_t *stream, const pb_field_t fields[], const void *src_struct)
                  ^
nanopb/pb_encode.c:675:18: note: code may be misoptimized unless -fno-strict-aliasing is used
```

This warning repeats for other functions as well in both the `pb_encode.c` and `pb_decode.c` files

PetteriAimonen, 2018-01-04:
Hi,

> When compiling nanbopb with the gcc-arm-none-eabi 2017q4 toolchain 
> (https://developer.arm.com/open-source/gnu-toolchain/gnu-rm) I get the 
> following warning when linking with LTO enabled:

Hmm, I wonder if it is the 'checkreturn' attribute that is causing it.

Can you try commenting out lines 14 and 16-18 of pb_encode.c, leaving 
only line 15:
#define checkreturn

And report back whether the warning still occurs?

--
Petteri

torfinnberset, 2018-01-04:
I just tried compiling with `#define checkreturn`, but it gives the same warning. 

Then, I tried removing all references to `checkreturn` completely, and it still gave this:

```
nanopb/pb_encode.h:164:6: warning: type of 'pb_encode_submessage' does not match original declaration [-Wlto-type-mismatch]
 bool pb_encode_submessage(pb_ostream_t *stream, const pb_field_t fields[], const void *src_struct);
      ^
nanopb/pb_encode.c:665:7: note: 'pb_encode_submessage' was previously declared here
 bool  pb_encode_submessage(pb_ostream_t *stream, const pb_field_t fields[], const void *src_struct)
       ^
nanopb/pb_encode.c:665:7: note: code may be misoptimized unless -fno-strict-aliasing is used
```

But then I looked more closely at all the compiler warnings, and noticed a different warning:

```
nanopb/pb.h:291:29: note: type 'struct pb_ostream_t' should match type 'struct pb_ostream_t'
 typedef struct pb_ostream_s pb_ostream_t;
                             ^
```

Since this struct is used in the function, I'm guessing that's related

PetteriAimonen, 2018-01-06:
Hmm, I tried to reproduce this issue but didn't get the warnings:

<pre>
petteri@oddish:~/nanopb$ ~/download/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc-7.2.1 -Wall -Wextra -fstrict-aliasing -flto -O2 -I. -c pb_encode.c pb_decode.c pb_common.c examples/simple/simple.c examples/simple/simple.pb.c

petteri@oddish:~/nanopb$ ~/download/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc-7.2.1 -Wall -Wextra -fstrict-aliasing -flto -O2 pb_encode.o pb_decode.o pb_common.o examples/simple/simple.o examples/simple/simple.pb.o -lnosys

petteri@oddish:~/nanopb$ arm-none-eabi-size -t a.out   text	   data	    bss	    dec	    hex	filename
  52585	   2492	    128	  55205	   d7a5	a.out
</pre>

What compiler params are you using?

PetteriAimonen, 2018-04-14:
I couldn't reproduce this warning so closing this off for now.

closed 2018-04-14 by PetteriAimonen

torfinnberset, 2018-08-28:
I found out there was something wonky with my build setup, causing ODR violations, it's gone now after cleaning up the build! My bad