nanopb/nanopb #306: Integer conversion resulted in truncation in nanopb.pb.c

Amomum, 2018-01-24:
I used simple .proto file:

    syntax = "proto2";
    import "proto/nanopb.proto"; 
    message R { 
        required uint32 date = 1;
        repeated uint32 test = 2 [(nanopb).max_count = 5];
    }

I than generated *.c and *.h for this .proto file and nanopb.pb.c and nanopb.pb.h from nanopb.proto and added them to my project.

If I use Keil 5 armcc compiler, I get this warning:

```
nanopb.pb.c(48): warning:  #69-D: integer conversion resulted in truncation
        
static const pb_field_t nanopb_fileopt_field = 
      PB_FIELD(1010, MESSAGE , OPTEXT, STATIC  , FIRST, nanopb_fileopt_struct, nanopb_fileopt, nanopb_fileopt, &NanoPBOptions_fields);
```

I looked in the preprocessed file and saw this:

```
static const pb_field_t nanopb_msgopt_field = 
      {1010, 0x00 | 0x00 | 0x07, 0, 0, (sizeof ((nanopb_msgopt_struct*)0)->nanopb_msgopt), 0, &NanoPBOptions_fields};
```

Scince first field of `pb_field_t` is `pb_size_t tag` and `pb_size_t` which is `unsigned char` that warning makes sense.

However I don't get this warning when compiling with gcc.

I'd like to know, if that's a bug or a spurious warning from armcc.

PetteriAimonen, 2018-01-24:
There are two things going on:

1) You shouldn't have / need nanopb.pb.c at all, because the generator excludes it by default. The reason why you are getting a reference to it is because your nanopb.proto is in a subdirectory. You can use e.g. `--nanopb_out=-xproto/nanopb.proto:output_directory` to exclude it from the includes.

2) The generated file should have a better error message for the large tag value, one that tells to define `PB_FIELD_16BIT` compilation option. However it looks like extension fields are currently not checked for this, which is a bug worth fixing.

Amomum, 2018-01-24:
That's interesting, thank you!
For simplicity, I just moved nanopb.proto from subdirectory and no nanopb.c is now needed. And all big numbers that caused a warning are gone.
I don't know why the came up in the first place but I guess it's still okay for me not to use `PB_FIELD_16BIT`.

It's also interesting why gcc did not show this warning even though I compiled with `-Wall -Wpedantic -Wextra`.

PetteriAimonen, 2018-01-24:
The big numbers come from nanopb.proto, so you should be fine now that it is not included on the C side.

GCC might have noticed that the file is actually never used. Normally it does give warnings about that I think.