nanopb/nanopb #307: Build error due to -fobjc-weak is not supported on the current deployment target

misaxi, 2018-02-03:
Not sure if this is related. But I'm developing an app using Ionic which targets iOS. The specific error I got described in the following issue.

https://github.com/phonegap/phonegap-plugin-push/issues/2183#issuecomment-362780912

PetteriAimonen, 2018-02-03:
Hmm, nanopb doesn't specify that build option so I assume it's coming from their build configuration.

hamath, 2018-02-03:
I don't know why this error suddenly happened, but fixed by the following links' way.
Let's try to update build target version in nanopb
https://stackoverflow.com/questions/36147625/xcode-7-3-cannot-create-weak-reference-in-file-using-manual-reference-counting
[image]

misaxi, 2018-02-04:
Thanks @hamath , I was able to build the project by setting the deployment target to iOS 9.0.

Looking forward to the official fixes.

Rdbrenna, 2018-03-30:
Click the Pods Project in Xcode(Usually at the bottom of the navigator) -> Under Targets, Click "nanopb" -> You will see that the Deployment Default is set to "4.3", Change this to the latest (11.3 for me) and it should be fixed :)

PetteriAimonen, 2018-04-14:
I think there is nothing to do here on the nanopb side, so I'm closing this off.

closed 2018-04-14 by PetteriAimonen

earthman1, 2018-04-25:
I have found that pod update resets the ios deployment target to 4.3. I added the following script to my podfile to reset the deployment target on each update:
```
post_install do |installer|
  installer.pods_project.targets.each do |target|
    target.build_configurations.each do |config|
      config.build_settings['IPHONEOS_DEPLOYMENT_TARGET'] = '9.0'
    end
  end
end
```
Of course, you may choose a different target than 9.0, by changing the target number in the above script.