nanopb/nanopb #308: Proto3 non-scalar (message) field lacking has_* in generated code

xanxys, 2018-02-06:
Proto3 scalar fields don't have "not-set" state, but message field does.
cf. [comments in official protobuf repository](https://github.com/google/protobuf/issues/249).

But generated struct didn't have any has_ field, so I can't specify "not-set" state for embedded message.
(I know I can work around this bug by using repeated w/ max_count=1 option, but it's not ideal.)

PetteriAimonen, 2018-02-06:
Ah, good point.

Makes sense to fix this. For now you can also use `(nanopb).proto3=false` on the single field to get a `has_` field.

Meyermagic, 2019-01-31:
I wrote a patch for this: https://github.com/Meyermagic/nanopb/commit/e319e5a3fc8cade64ce217d62d460d2c6d15b560

This breaks the alltypes_proto3 test, not sure if any new tests should be introduced. I'll send a PR some time next week with the test fix and any other suggestions you have (doc changes? does an option need to be introduced to get back the old behavior for cases where the extra bytes makes the generated structs prohibitively large?)

Thanks for the great library!

PetteriAimonen, 2019-02-01:
I think it will need a generator option or something to help the transition. It will break compatibility for a lot of code otherwise.

adamlepkowski, 2019-03-22:
Are you going to solve this problem?

PetteriAimonen, 2019-03-22:
@adamlepkowski Eventually, sure. But as usual, lack of time etc.

PetteriAimonen, 2019-11-11:
Proposed fix is now on dev_proto3_msgs branch for comments, discuss here or on the forum:
https://groups.google.com/forum/#!topic/nanopb/En4THnRSNWo

A bit annoying about the breaking change, but I've been postponing this for too long already so better just do it now.

PetteriAimonen, 2019-12-16:
Merged already to master.

closed 2019-12-20 by PetteriAimonen