nanopb/nanopb #319: Option to remove "automatically generated" comment in generated file

Amomum, 2018-03-27:
I added custom build step to my build system that regenerates .c and .h files for my .proto file.
Unfortunately, nanopb puts this comment with current time and date in them :
```
/* Automatically generated nanopb constant definitions */
/* Generated by nanopb-0.3.9 at Mon Mar 26 19:15:47 2018. */
```
That makes git sad because files change on every build.

Is it possible to stop this comment from generating?

PetteriAimonen, 2018-03-27:
Yeah, use option `-T`, for example:

<pre>
nanopb_generator.py -T ...
</pre>

or if used through protoc:

<pre>
protoc --nanopb_out=-T:outdir ...
</pre>

Amomum, 2018-03-27:
Oh, that's great!
I searched in the docs but didn't find that option there. Is it mentioned there?

PetteriAimonen, 2018-03-27:
I guess not; -h to the generator shows some help.

closed 2024-09-19 by PetteriAimonen