nanopb/nanopb #322: Missing documentation for bytes defaults

maxshaffer, 2018-04-06:
Nanopb documentation does not indicate how to set default values in the proto file for a field of type 'bytes'.

For example:
message testbytes {
   required bytes trial = 1 [default = "\xDEADBEEF", (nanopb).max_size = 4, (nanopb).fixed_length=true];
}

Expected result:
{{0xDE, 0xAD, 0xBE, 0xEF}}

Actual result:
{{0x5c,0x33,0x33,0x36,0x41,0x44,0x42,0x45,0x45,0x46}}

PetteriAimonen, 2018-04-06:
Seems like a duplicate of issue #316. Can you try newest git version?

maxshaffer, 2018-04-06:
Just pulled down the latest updates. I still get the same incorrect result.

PetteriAimonen, 2018-04-07:
Indeed, looks like fixed_length option is still broken.

PetteriAimonen, 2018-04-07:
I added some notes to docs also, https://jpa.kapsi.fi/nanopb/docs/concepts.html#default-values

The correct syntax to use is like \xDE\xAD\xBE\xEF, same as in C.