nanopb/nanopb #323: when i decode a buffer use pb_decode, i get a string overflow error msg

xiaobaxiang, 2018-04-14:
first I serialize an protobuffer object ,get a arraybuff And then copy it into another array.
when i deserialize from the array with pb_decode,i get  a string overflow error msg

here is the protofile:
syntax = "proto3";
package SocketCmd;

message CmdHeader {
	string cmdCode = 1;
	string identity = 2;
	string oppositeId = 3;
	string timeToken = 4;
	int32 resultCode = 5;
	uint32 serverId = 6;
}

PetteriAimonen, 2018-04-14:
"string overflow" would mean that you have specified a length for one of the string fields (probably in your .options file), and the string you are trying to pass is longer than this. Note that `max_size` must be long enough to contain the C string null terminator also.

xiaobaxiang, 2018-04-17:
thank for your reply.the problem seems  caused by null terminator.I have a doubt that the arraybuff  of  serialization does not contain the  C string null terminator,is that right?

PetteriAimonen, 2018-04-17:
The serialized message does not have to contain the terminator, nanopb adds it automatically when decoding.

You could try decoding your data with e.g. https://protogen.marcgravell.com/decode and see what length the string there is, compared to the setting in your .options.

xiaobaxiang, 2018-04-17:
Thank you so much

closed 2018-04-17 by xiaobaxiang