nanopb/nanopb #324: nanopb_generator should include google/proto/descriptor.pb.h in generated protos that import descriptor.proto

seanmc602, 2018-05-10:
When you import non-standard proto messages into another, the generator includes those generated headers into the other. However, when you import descriptor.proto, the #include line doesn't make it.

PetteriAimonen, 2018-05-10:
Hmm, yeah, it is currently hardcoded in excludes on line 1596 of nanopb_generator.py. The idea is that descriptor.proto is usually included to define custom field options, but nanopb doesn't do anything with those anyway (because it has no reflection support).

I guess that at a minimum, it would make sense to have a command line option to override the default excludes.

PetteriAimonen, 2019-12-20:
There is now a generator option `include` that can be used to include arbitrary files into the generated `.pb.h` files.

This should be usable as a workaround for this bug also. I think I still want to exclude the descriptor.pb.h by default as for most code that is not used on C side.

closed 2019-12-20 by PetteriAimonen