nanopb/nanopb #325: const qualifier applied to a wrong place in pb_callback_s.encode

edyfox, 2018-05-10:
In the previous version of nanopb, `pb_callback_s.encode()` took a point as the third argument. The pointer points to a read-only memory block of arbitrary type:

https://github.com/nanopb/nanopb/blob/master/pb.h#L298

In the latest version, in order to support modifying the pointer inside the callback, we changed the third argument to a pointer-to-pointer type. In order to server such intention, this pointer should be a pointer pointed to a (modifiable) pointer, which points to a read-only memory block of arbitrary type. It should be translated to code as:

```
  const void** args
```

However, the current code put `const` at a wrong place:

https://github.com/nanopb/nanopb/blob/master/pb.h#L304

So the actual meaning of that arg becomes: this is a pointer pointed to a non-modifiable pointer, which points to a memory block that can be modified.

Unfortunately, the fix to this problem is going to an API-breaking change. We'll need a flag to allow the users of nanopb to keep the current (wrong) behavior of that function, so they can have some time to upgrade their code.

PetteriAimonen, 2018-05-11:
I think this is actually intentional.

Because `pb_encode()` takes a const pointer to the destination struct, it gives a pointer to const pointer as the argument to the callback. This way nanopb itself will not cause the structure being encoded to be accidentally modified. If the user code wishes to do so, it can cast away the const.

edyfox, 2018-05-11:
OK, in that way, at least it should be `const void *const *args`, to be consistent with the previous (deprecated) interface.

closed 2018-11-10 by PetteriAimonen