nanopb/nanopb #328: JSON Support

kartur, 2018-06-19:
I would like to humbly request a feature for encoding/decoding JSON. :)

PetteriAimonen, 2018-06-19:
Yeah, would make sense to have.

Some prior discussion in issue #159.

Sreezz, 2020-05-11:
Is  json mapping possible with nanopb.

PetteriAimonen, 2020-05-11:
No, there is no implementation and no ongoing work on it currently.

mhaberler, 2023-05-11:
Hi Petteri,

I would be interested in serializing to JSON, too - I see this is a low-prio issue cooking for a while now

I might be able to put some time into it (given I'm an MBA this might not mean much in results ;) - early stage thought warnings apply

I can envisage two routes in my use case:

a) try to serialize off to JSON given an already encoded buffer
b) try to serialize off to JSON off the generated C message structs

b) might sound whacky but given introspection is missing in nanopb:

what about applying the trick below to https://github.com/nanopb/nanopb_cpp ?

it doesnt have to be C-only for me, and at least it could be a start

what do you think, is this viable?

Michael

https://stackoverflow.com/questions/17549906/c-json-serialization - accepted answer
http://coliru.stacked-crooked.com/a/a5902693e51b59b9 - code example

ps: good to see nanop growing in adoption and being a stable component - great job!

PetteriAimonen, 2023-05-11:
JSON serialization is quite straightforward, it could be done either by generating a function that directly prints stuff, or by using the descriptor mechanism from `pb_common.h`. Deserialization is a bit harder, but something like jsmn library could help.

One wacky approach would be to generate a JSON serializer function on demand based on the `_FIELDLIST` macro.
Here is a simple demo:

<pre>
#include "simple.pb.h"
#include &lt;pb_encode.h>
#include &lt;stdio.h>

/* These macro definitions and helper functions would go into
 * pb_json.c/h in actual implementation */

bool pb_json_write_fieldname(pb_ostream_t *stream, const char *fieldname, bool *first_field)
{
    // Print field separator comma
    if (!*first_field)
    {
        if (!pb_write(stream, ", ", 2)) return false;
    }
    *first_field = false;
    
    // Print field name and colon
    return pb_write(stream, "\"", 1) &&
           pb_write(stream, fieldname, strlen(fieldname)) &&
           pb_write(stream, "\": ", 3);
}

bool pb_json_write_int32(pb_ostream_t *stream, int32_t value)
{
    char buf[16];
    int n = snprintf(buf, sizeof(buf), "%d", value);
    return pb_write(stream, buf, n);
}

#define PB_JSON_SERIALIZER(msgname, structname) \
    bool msgname ## _write_json(pb_ostream_t *stream, const structname *src) \
    { \
        bool first_field = true; \
        return pb_write(stream, "{", 1) && \
               msgname ## _FIELDLIST(PB_JSON_GEN_SERIALIZER, structname) \
               pb_write(stream, "}", 1); \
    }

#define PB_JSON_GEN_SERIALIZER(structname, atype, htype, ltype, fieldname, tag) \
    pb_json_write_fieldname(stream, #fieldname, &first_field) && \
    PB_JSON_WRITE_ ## atype ## _ ## htype ## _ ## ltype(fieldname) &&

#define PB_JSON_WRITE_STATIC_REQUIRED_INT32(fieldname) pb_json_write_int32(stream, src->fieldname)

/* This is example main program */

// This macro call generates JSON serialization function with following prototype:
bool SimpleMessage_write_json(pb_ostream_t *ostream, const SimpleMessage *src);
PB_JSON_SERIALIZER(SimpleMessage, SimpleMessage);

int main()
{
    SimpleMessage msg = {};
    msg.lucky_number = 12345;
    
    uint8_t buf[256];
    pb_ostream_t stream = pb_ostream_from_buffer(buf, sizeof(buf));
    
    if (!SimpleMessage_write_json(&stream, &msg))
    {
        printf("JSON serializer failed\n");
        return 1;
    }
    else
    {
        buf[stream.bytes_written] = '\0'; // Null-terminate string
        printf("JSON result: %s\n", (char*)buf);
        return 0;
    }
}
</pre>

<pre>
petteri@raichu:~/nanopb/examples/simple$ gcc -o simple_json simple_json.c simple.pb.c  ~/nanopb/pb_encode.c ~/nanopb/pb_common.c -I ~/nanopb
petteri@raichu:~/nanopb/examples/simple$ ./simple_json 
JSON result: {"lucky_number": 12345}
</pre>

mhaberler, 2023-05-11:
thanks, I will give this a spin using [ArduinoJson](https://arduinojson.org/) and post what I can find out!

Michael

lobis, 2024-01-05:
I would also love to have the built-in JSON serialization (I also don't care for deserialization).