nanopb/nanopb #331: Help creating repeated submessages

umanayana, 2018-08-17:
Hello, 

I need to create a submessage if "logs" which are variable in length and then that submessage need to be held within another message which can tell me the number of logs I have. I then need to pack all this data into buffer which cannot exceed 512 bytes. 

**Here is the Proto file so far**
`message mqtt_log_pl_t 
{
   repeated bytes serial 	= 1;
   repeated bytes logs = 2;  
}

message mqtt_logs_pl_t 
{
	required uint32 log_count = 1;
	required mqtt_log_pl_t logs = 2;
}`

**mqtt_log_pl_t**

* This message holds a single log message.
* `log`: is a byte array of varying size. I am think of putting this into it's own messages and serialise it using `pb_encode_delimited()` so I can encode the size with it.
* `serial`: Fixed length of 4 bytes. But I don't specify it as fixed because I want to use the callback to get the serial number and encode directly into buffer.

**mqtt_logs_pl_t**

* Holds an array of `mqtt_log_pl_t` message depends on how many I can fit into my 512 byte buffer.
* `log_count`: Intended to indicate the total number of `log` which were encoded? I am not sure about this but I need some way to know how many log messages were encoded. Is there a better way to do this.

umanayana, 2018-08-18:
@PetteriAimonen 
After a bit of Googling and from what you said I am thinking this.

On each callback to `mqtt_logs_pl_t` just encode 1 `mqtt_log_pl_t` then return check the encoded size and do it again as long as we have space in the buffer and more message to encode.

```
bool CallBack_mqtt_logs_pl_t(pb_ostream_t *stream, const pb_field_t *field, void * const *arg)
{
	/*Setup mqtt_log_pl_t callbacks 
	Encode 1 mqtt_log_pl_t and return*/
	pb_encode_delimited(stream, mqtt_log_pl_t_field, &current_log)
}

Encode_As_Many_Logs_As_Possible
{
	/*Initialize the output stream here*/

	do
	{
		uint8_t data[max_msg_size];

		size_of_log_message = Get_1_Log_Message(&data);

		mqtt_logs_pl_t logs
		/*Setup the callback and data for logs*/

		pb_encode(stream, mqtt_logs_pl_t_fields, &logs);

	}while( !is_log_buffer_empty && current_stream_size < 512 )

	//Send the encoded buffer here
}
```

PetteriAimonen, 2018-08-18:
pb_encode_delimited() doesn't work inside callbacks, instead you should use pb_encode_tag() and then pb_encode_submessage().

Though if you need particularly tight control of memory usage and the encoding process, you can bypass pb_encode() completely. Instead doing something like:

```
pb_encode_tag(stream, PB_WT_STRING, mqtt_log_pl_t_serial_tag);
pb_encode_bytes(...);
pb_encode_tag(stream, PB_WT_STRING, mqtt_log_pl_t_logs_tag);
pb_encode_bytes(...);
```

and then wrapping that in a submessage if that is the structure you want. The protobuf encoding spec will help building the correct structure, which is the same whether you use callbacks from pb_encode() or just call the functions directly: https://developers.google.com/protocol-buffers/docs/encoding

umanayana, 2018-08-19:
I took a look at submessage encoding that won't work. When I read the data if I don't encode it and store it in the 512 byte buffer I won't have anywhere else to store it. So I have to read the data and encode on the fly. 

I don't fully understand your second approach. Could you provide a better example? 

All I want to do is encode as many `log` messages as I can into a 512 byte buffer. A single log message looks like `mqtt_log_pl_t`. All of these method need to use the callback method to serialize the data due to memory constraints. 

Is there better example of how this can be done? As I'm not following your suggestions. 

Thanks

PetteriAimonen, 2018-08-20:
I don't currently have enough hobby time available to make a more complete example.

But you can work it out by e.g. encoding the data you want with some test code, and then compare the output of your more memory-efficient version until you get it to match.

umanayana, 2018-08-20:
@PetteriAimonen 
That understandable I am having some trouble generating some code though . I am currently testing out different things but I ran into a problem where the code is not generating correctly.

Using the .proto and .options below the generated C struct for `mqtt_log_pl_t` used `pb_callback_t`. How can I get it to declare these as arrays.

```
message mqtt_log_pl_t 
{
	repeated bytes 	serial 		= 1;
	repeated string log_str		= 2;  
}
```

```
mqtt_log_pl_t.serial 		max_size:9
mqtt_log_pl_t.serial 		fixed_length:true

mqtt_log_pl_t.log_str 		max_size:129
mqtt_log_pl_t.log_str 		fixed_length:true
```

The .proto and .options above generate the C code below

```
PB_PACKED_STRUCT_START
typedef struct _mqtt_log_pl_t {
    pb_callback_t serial;
    pb_callback_t log_str;
/* @@protoc_insertion_point(struct:mqtt_log_pl_t) */
} pb_packed mqtt_log_pl_t;
PB_PACKED_STRUCT_END
```

PetteriAimonen, 2018-08-20:
That's because those are repeated fields, so to make them into (two-dimensional) arrays, you need to specify `max_count` also.

umanayana, 2018-08-20:
Thanks for the help. I've managed to get it to work as I needed.

closed 2018-11-10 by PetteriAimonen