nanopb/nanopb #334: Help decode a repeated byte field

umanayana, 2018-08-28:
I am trying to decode a repeated byte field (below) without copy the data anywhere. I just need to know where the data begins in the raw input buffer.

I cannot copy the data out of the buffer due to memory constraints so I need to work with the raw input buffer. I will be using the decode callback but I don't know how to figure out where the binary data begins.

What's the best way to achieve this?

```
message message 
{
	repeated bytes	binary = 1;	
}
```

umanayana, 2018-08-28:
Just an update I have done the following after taking a look at how `pb_read()` and `buf_read()` works. This code snippet runs from the callback.

To my understanding if I set stream->bytes_left to zero the callback will only run once.

```
 otafw_arg->binary_size = stream->bytes_left;

/*Manually manipulate the stream to find the location of the binary data
 * @ref buf_read() from nanopb API*/
 otafw_arg->binary   = (uint8_t *)stream->state;
stream->state       = (pb_byte_t *)stream->state + stream->bytes_left;
stream->bytes_left  = 0; /*Stop the callback from running again*/
```

PetteriAimonen, 2018-08-29:
Yeah, your solution works as long as you only use buffer inputs.

umanayana, 2018-08-29:
@PetteriAimonen 

> only use buffer inputs.

What you do mean by this?

The message will actually look as below. When I encode I am careful to encode in field tag order. That is the `binary` field is always at the end of the encode buffer. 
If I use the method above to read the `binary` field, will this cause any potential problems?

```
enum messages_type 
{
	A	=1; 
	B	=2;
	C	=3;
}

message message 
{
	required messages_type type 	= 1;

	oneof u
	{
		uint32 size 	= 2;
		uint32 count 	= 3;
	}
		
	repeated bytes	binary 			= 4;	
}
```

PetteriAimonen, 2018-08-29:
I just meant that it only works with streams from `pb_istream_from_buffer()`.

closed 2018-11-10 by PetteriAimonen