nanopb/nanopb #336: Problem with incorrect processing of certain messages containing bytes arrays

vrleto, 2018-09-06:
Hi,

I'm trying to solve some problem and run out of ideas. I hope you could help me solve that problem. Currently I'm using your newest release package v0.3.9.1 and compile my test.proto file with protoc included in your binary release. All development is done on 64bit ubuntu 16.04 with gcc 5.4.0 . Problem is that I can't make nanopb to correctly process size of Guid message (it consist 16 byte array and 32bit size (as PB_FIELD_32BIT is on)  but nanopb decoder for some reason seems to take size from incorrect place in stream resulting in "bytes size exceeded" error message as shown bellow). Any help would be greatly appreciated. Included all necessary information and test code:

**test.proto**:
syntax = "proto2";
enum Event {
    NoEvent = 0x00;
    BeginSessionEvent = 0x01;                 
}

message BeginSession {
    required Guid SystemGuid = 1;              
    required Guid SessionGuid = 2;            
    required uint64 Time = 3;                  
    required sint32 TimeZoneBias = 4;          
    required SessionFlags Flags = 5;           
    required uint32 SensorVersion = 6;         
}

enum SessionFlags {
    NoSessionFlags = 0x00;
    BootSession = 0x01;                       
}

message Guid {
    required bytes Data = 1;
}

**test.options**:
Guid.Data                                  max_size:16

**test.c**:
#include <stdio.h>
#include <stdlib.h>
#include <ctype.h>

#define PB_ENABLE_MALLOC 1
#define PB_FIELD_32BIT 1 

#ifdef PB_ENABLE_MALLOC
void *ProtobufRealloc(void *ptr, size_t newsize);
void ProtobufFree(void *ptr);
#define pb_realloc(ptr, size) ProtobufRealloc(ptr, size)
#define pb_free(ptr) ProtobufFree(ptr)
#endif

#include "pb_encode.h"
#include "pb_decode.h"
#include "test.pb.h"

int main(int argc, char **argv) {
    char buffer[4096];
    size_t size;
    BeginSession session = {0};
    pb_ostream_t stream;

    memset(&session.SystemGuid.Data.bytes, 'b', 16);
    session.SystemGuid.Data.size = 16;
    memset(&session.SessionGuid.Data.bytes, 'a', 16);
    session.SessionGuid.Data.size = 16;

    session.Time = 10000;
    session.TimeZoneBias = 100;
    session.Flags = 0;
    session.SensorVersion = 10;

    if (!pb_get_encoded_size(&size, BeginSession_fields, &session)) {
        printf("error\n"); 
    }

    stream = pb_ostream_from_buffer(buffer, 4096);

    if (!pb_encode(&stream, BeginSession_fields, &session)) {
        printf("error: %s\n", PB_GET_ERROR(&stream));
    }
    printf("size beginsession %d\n", (int)size);
}

result:
error
error: bytes size exceeded
size beginsession 0

As you can see there is something wrong with interpretation of size of Guid (session.SystemGuid.Data.size) - while debugging function pb_enc_bytes() from pb_encode.c I noticed that bytes->size was way to high and equal to value of  'b' so it looks like size got somehow overwritten or read from wrong location even that session.SystemGuid.Data.bytes arrays has size equat to 16 so that shouldn't be possible. Please let me know what is wrong here. Thank you.

PetteriAimonen, 2018-09-07:
You can't set these options in `test.c`:

```
#define PB_ENABLE_MALLOC 1
#define PB_FIELD_32BIT 1
```

because then they will only affect your own program and not the code in `pb_decode.c`. Instead, you can either set them by editing `pb.h`, or on the compiler command line / makefile / project config.

vrleto, 2018-09-07:
Yes you are right! - I have added -DPB_ENABLE_MALLOC  -DPB_FIELD_32BIT to Makefile and it works now as expected! Thank you for your help in solving that issue, appreciate that.

closed 2018-09-07 by vrleto