nanopb/nanopb #337: encode using nanopb and decode using golang - unexpected EOF

sunilvaltix, 2018-09-07:
We have a backend server process which writes buffers encoded using nanopb into shared memory and another process reads from shared memory, decodes it into a go struct and sends it over gRPC to a controller.

The backend server is written in C, and the agent process which talks to north facing controller is written in go. My understanding was that the encoded stream would be decodable on go side. Is that not the case? Is this supposed to work?

I have hex dumped the stream on both sides to make sure that its not the (shared memory) infra in between that's the culprit. The stream being decoded is same as the encoded stream generated by nanopb.

PetteriAimonen, 2018-09-08:
Yes, any protobuf implementation should be able to decode stream from any other implementation.

sunilvaltix, 2018-09-08:
This turned out to be a bug in the go code that was doing the unmarshal. Closing.

closed 2018-09-08 by sunilvaltix