nanopb/nanopb #338: compiler memory usage

Cloudef, 2018-09-13:
Since the nanopb compiler writes everything into memory before streaming it to protoc, the memory usage can climb quite fast. Sometimes even so much that it's impossible to compile .proto specifications with lots of dependencies (without lots of RAM of course). The most contributing factor to the output size seems to be the EncodedSize class with its `symbols` member (which is later joined into string in the `__str__` method), which can grow really big for oneof fields that contain lots of members.

Since I can't change the proto files itself that I'm generating from, nor take out the bloat. I'm currently workarounding this by not generating the `_size` define at all (which we don't use anyways), that causes memory usage to spike for fields which size can't be determined offline.

To fix this problem, I think instead of formatting strings and storing the sources/headers in variables, the whole generator should be designed as something that takes in the stream and produces stream, that gets written directly to the stdout. While there are already lots of methods using `yield`, it's basically pointless in the end when the result is stored in single variable.

I'm not familiar with protoc internals or how it communicates with its plugins, so I'm not sure if this is even possible though.

PetteriAimonen, 2018-09-13:
Making it fully streaming could be a quite large job.

Could you provide some statistics of your use case? I.e. how large in total are the input .proto files, how large in total the .pb.c and .pb.h output files, and what is the maximum memory usage you see?

Cloudef, 2018-09-13:
Sadly the proto files itself are under NDA. But the generator happens to spike out all the 7GB of memory I have allocated for the VM it's running in.

The proto file itself contains oneof with fairly complex submessages, so there's most likely heavy nesting involved.

Example:
```
message buffer {
    repeated Event events = 1;
}

message Event {
    oneof event {
        Foo foo = 1;
        Bar bar = 2;
        ... lots of these ...
    }
}
```

The issue may fix itself eventually, if I fine-tune all the dependencies with `.options` file, as I think many of the dynamic fields could have maximum length. Even though the generator wouldn't run out of memory, it would surely generate pretty ugly file for this particular proto specification (with tons of symbols appended after the `_size` define).

Cloudef, 2018-09-13:
The input proto files range between 1.1K-5K. The output for this particular file without generating the `_size` define is 6.2K. I don't even want to imagine the size if I would generate it with machine with more ram... 😄

PetteriAimonen, 2018-09-13:
Ok, then this sounds definitely like a bug. Probably something is creating a circular reference or something.
I wouldn't expect a 7 GB memory usage unless you have hundreds of megabytes worth of .proto files.

I'll see if I can figure out a way to reproduce the behaviour.

Cloudef, 2018-09-13:
The issue goes away if I comment out enough messages from the oneof, I'll see if I can give some files that reproduce the issue as well.

Cloudef, 2018-09-13:
Is there something wrong doing it like this? Does the ternary operations serve a purpose?

```diff
diff --git a/cpp/protobuf/nanopb/generator/nanopb_generator.py b/cpp/protobuf/nanopb/generator/nanopb_generator.py
index 32b8c669..d9d897d9 100755
--- a/cpp/protobuf/nanopb/generator/nanopb_generator.py
+++ b/cpp/protobuf/nanopb/generator/nanopb_generator.py
@@ -856,13 +856,7 @@ class OneOf(Field):
             elif size.value > largest.value:
                 largest = size
 
-        if not symbols:
-            return largest
-
-        symbols = list(symbols)
-        symbols.append(str(largest))
-        max_size = lambda x, y: '({0} > {1} ? {0} : {1})'.format(x, y)
-        return reduce(max_size, symbols)
+        return largest
 
```

Cloudef, 2018-09-13:
Here is a example .proto file that reproduces this. If not, either reduce ram, or generate more "BigValue"'s 😄 

```
syntax = "proto3";

package enterprise;

message BigData {
   repeated BigEvent events = 1;
}

message BigEvent {
   oneof event {
      BigValue1 data1 = 1;
      BigValue2 data2 = 2;
      BigValue3 data3 = 3;
      BigValue4 data4 = 4;
      BigValue5 data5 = 5;
      BigValue6 data6 = 6;
      BigValue7 data7 = 7;
      BigValue8 data8 = 8;
      BigValue9 data9 = 9;
      BigValue10 data10 = 10;
      BigValue11 data11 = 11;
      BigValue12 data12 = 12;
      BigValue13 data13 = 13;
      BigValue14 data14 = 14;
      BigValue15 data15 = 15;
      BigValue16 data16 = 16;
      BigValue17 data17 = 17;
      BigValue18 data18 = 18;
      BigValue19 data19 = 19;
      BigValue20 data20 = 20;
      BigValue21 data21 = 21;
      BigValue22 data22 = 22;
      BigValue23 data23 = 23;
      BigValue24 data24 = 24;
      BigValue25 data25 = 25;
      BigValue26 data26 = 26;
      BigValue27 data27 = 27;
      BigValue28 data28 = 28;
      BigValue29 data29 = 29;
      BigValue30 data30 = 30;
      BigValue31 data31 = 31;
      BigValue32 data32 = 32;
   }
}

message BigValue1 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue2 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue3 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue4 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue5 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue6 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue7 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue8 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue9 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue10 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue11 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue12 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue13 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue14 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue15 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue16 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue17 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue18 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue19 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue20 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue21 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue22 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue23 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue24 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue25 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue26 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue27 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue28 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue29 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue30 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue31 {
   int32 key = 1;
   bytes value = 2;
}

message BigValue32 {
   int32 key = 1;
   bytes value = 2;
}
```

PetteriAimonen, 2018-09-13:
Ah, this seems like an unexpected side effect of commit f8938774

PetteriAimonen, 2018-09-13:
Heh, yeah, it doubles the memory usage for each value. With 32 values it's trying to generate 186 gigabytes of output.

I'll see if it is possible to compute this at runtime more efficiently, otherwise I'll just add some limit to it.

PetteriAimonen, 2018-09-13:
Should be fixed now.

Cloudef, 2018-09-13:
Confirmed. Closing

PetteriAimonen, 2018-09-13:
I'll just keep this open until fix is released, in case someone else stumbles into it they can find it easier.

PetteriAimonen, 2018-11-10:
Fix released in 0.3.9.2.

closed 2018-11-10 by PetteriAimonen