nanopb/nanopb #339: the generated C# source code refers to global::NanopbReflection.Descriptor

xiaobaxiang, 2018-09-18:
The C # source generated using the proto file references global::NanopbReflection.Descriptor, where can I find global::NanopbReflection.Descriptor

PetteriAimonen, 2018-09-18:
You can compile `nanopb/generator/proto/nanopb.proto` using the C# generator to get that file.

xiaobaxiang, 2018-09-18:
C# generator  only supports proto3 syntax，but the nanopb.proto uses proto2 syntax

xiaobaxiang, 2018-09-18:
I edit the proto2 syntax nanopb.proto to proto3,is this available?

PetteriAimonen, 2018-09-18:
Dunno if it will work that way.

But another option is to put the nanopb options in separate `myproto.options` file so that the C# generator does not see them.

xiaobaxiang, 2018-09-18:
how to separate?which file?Is the nanopb/generator/proto/nanopb.proto?It's not under the folder generator-bin.

xiaobaxiang, 2018-09-18:
The commonjs style js source generated using the proto file also require nanopb_pb.js.I don't konw how to get the nanopb_pb.js.

PetteriAimonen, 2018-09-18:
You can use the separate `.options` file like this example:
https://github.com/nanopb/nanopb/blob/master/examples/network_server/fileproto.options

That way you don't need to reference nanopb.proto in the .proto file and it is easier to use with other libraries.

xiaobaxiang, 2018-09-19:
Which folder should I put myproto.options in?What's in the content?Is the content in nanopb.proto?If I don't refer to nanopb.proto, does [(nanopb).max_size = 5] need to be deleted?

PetteriAimonen, 2018-09-19:
See https://jpa.kapsi.fi/nanopb/docs/reference.html#defining-the-options-in-a-options-file

xiaobaxiang, 2018-09-19:
Thanks a lot, I've solved it.

closed 2018-09-19 by xiaobaxiang