nanopb/nanopb #342: Possible null-pointer dereference in pb_decode.c

Amomum, 2018-10-03:
Here it is:

``` cpp
static bool checkreturn decode_callback_field(pb_istream_t *stream, pb_wire_type_t wire_type, pb_field_iter_t *iter)
{
    pb_callback_t *pCallback = (pb_callback_t*)iter->pData;
    
#ifdef PB_OLD_CALLBACK_STYLE
    void *arg = pCallback->arg;
#else
    void **arg = &(pCallback->arg);
#endif
    
    if (pCallback == NULL || pCallback->funcs.decode == NULL)
return pb_skip_field(stream, wire_type);
```

`pCallback->arg` is dereferenced before `pCallback` is checked against NULL. This can potentially lead to undefined behaviour.

Not sure how bad it actually is.

PetteriAimonen, 2018-10-03:
Hmm, good catch. This is certainly undefined behaviour per the C standard, especially with PB_OLD_CALLBACK_STYLE.

PetteriAimonen, 2018-10-03:
Fixed now. Looks like the impact is not particularly high because it can occur only in very specific situations, when all of the following apply:

1) Message contains extension fields and
2) the extension field is of callback type and
3) the pb_extension_t structure is defined, but the pointer to pb_callback_t is left as NULL.

The correct behaviour in this case is that the field will be skipped.

Further, for most compilers atleast and probably for all compilers, the
bug will only occur if PB_OLD_CALLBACK_STYLE compilation option is specified.

roman-khazanskii, 2018-10-03:
All credits is due to PVS-Studio, not me :)

Amomum, 2018-10-03:
Opps, wrong account.
Anyway, all credit is due to PVS-Studio, not me :)

PetteriAimonen, 2018-11-10:
Fix released in 0.3.9.2.

closed 2018-11-10 by PetteriAimonen