nanopb/nanopb #343: nanopb_generator.py doesn't conform with rule_protobuf, proto_compile form for specifying options

seanmc602, 2018-10-04:
When using the proto_compile rule from https://github.com/pubref/rules_protobuf, you can use the pb_options attribute to pass "command-line" options to the plug-in, nanopb_generator.py. However, these options are passed as a CSV list instead of a quoted, space-delimited list as one would use in the command-line version:

```
protoc --descriptor_set_out=example_proto_c.descriptor_set --plugin=protoc-gen-nanopb-lang=protoc-gen-nanopb --nanopb-lang_out=--no-timestamp,--extension=.nanopb,--options-path=mypath:output_path --proto_path=mypath --proto_path=. mypath/example.proto
```

Throws this error:
```
Usage: nanopb_generator.py [options] file.pb ...

protoc-gen-nanopb: error: no such option: --no-timestamp,--extension
```

If you quote the options and space them, it is happy:
```
protoc --descriptor_set_out=example_proto_c.descriptor_set --plugin=protoc-gen-nanopb-lang=protoc-gen-nanopb --nanopb-lang_out="--no-timestamp --extension=.nanopb --options-path=mypath":output_path --proto_path=mypath --proto_path=. mypath/example.proto
```
but that doesn't comply with the way other plugins accept options when using the protoc_plugin form.

PetteriAimonen, 2018-10-04:
Ok, seems reasonable to allow comma separators.

PetteriAimonen, 2019-03-07:
Note to self: fix backported to 0.3.9.x

PetteriAimonen, 2019-03-08:
Fix released in nanopb-0.3.9.3.

closed 2019-03-08 by PetteriAimonen