nanopb/nanopb #349: Decode callback cannot be run

DenisArmy, 2018-10-16:
Hello all,

I'm a biginner in protobufs.
I have an issue with repeated fields decoding.

    // set callbacks
    message.msg1.msg2.funcs.decode = &comms_pb_decode_arr_callback;

    // decode
    bool status = pb_decode(&stream, fields, &message);

But callback wasn't called.

According my investigation I found the following code in pb_decode.c:
static bool checkreturn decode_static_field(pb_istream_t *stream, pb_wire_type_t wire_type, pb_field_iter_t *iter)
{
...
        case PB_HTYPE_ONEOF:
            *(pb_size_t*)iter->pSize = iter->pos->tag;
            if (PB_LTYPE(type) == PB_LTYPE_SUBMESSAGE)
            {
                /* We memset to zero so that any callbacks are set to NULL.
                 * Then set any default values. */
                memset(iter->pData, 0, iter->pos->data_size);
                pb_message_set_to_defaults((const pb_field_t*)iter->pos->ptr, iter->pData);
            }
            return func(stream, iter->pos, iter->pData);
...
}

Is any a common way to get arround this problem?

Thanks in advance.

PetteriAimonen, 2018-10-16:
Seems like a duplicate of #175: Callbacks inside oneofs are currently not supported. Problem is that if the oneof is stored as C union, the callbacks between different fields would occupy same memory area.

Workarounds include specifying `(nanopb).no_unions = true`, or using a callback for the oneof field also.

DenisArmy, 2018-10-16:
Thanks. Sorry for duplication.

DenisArmy, 2018-10-16:
Should I add [ (nanopb).no_unions = true ] for one of field or for repeated field?

PetteriAimonen, 2018-10-16:
To oneof field.

DenisArmy, 2018-10-16:
Looks like it still generate c union.

proto file: 
`syntax = "proto2";

import "nanopb.proto";

message ThirdMessage {
    required bool test = 1;
}

message SecondMessage {
    repeated bool test = 1;
} 

message FirstMessage {
    oneof datails {
        SecondMessage second = 1 [(nanopb).no_unions = true];
        ThirdMessage third = 2 [(nanopb).no_unions = true];
    }
}`

header:
`/* Automatically generated nanopb header */
/* Generated by nanopb-0.4.0-dev at Tue Oct 16 17:56:36 2018. */

#ifndef PB_SIMPLE_PB_H_INCLUDED
#define PB_SIMPLE_PB_H_INCLUDED
#include <pb.h>

/* @@protoc_insertion_point(includes) */
#if PB_PROTO_HEADER_VERSION != 30
#error Regenerate this file with the current version of nanopb generator.
#endif

#ifdef __cplusplus
extern "C" {
#endif

/* Struct definitions */
typedef struct _SecondMessage {
    pb_callback_t test;
/* @@protoc_insertion_point(struct:SecondMessage) */
} SecondMessage;

typedef struct _ThirdMessage {
    bool test;
/* @@protoc_insertion_point(struct:ThirdMessage) */
} ThirdMessage;

typedef struct _FirstMessage {
    pb_size_t which_datails;
    union {
        SecondMessage second;
        ThirdMessage third;
    } datails;
/* @@protoc_insertion_point(struct:FirstMessage) */
} FirstMessage;

/* Default values for struct fields */

/* Initializer values for message structs */
#define ThirdMessage_init_default                {0}
#define SecondMessage_init_default               {{{NULL}, NULL}}
#define FirstMessage_init_default                {0, {SecondMessage_init_default}}
#define ThirdMessage_init_zero                   {0}
#define SecondMessage_init_zero                  {{{NULL}, NULL}}
#define FirstMessage_init_zero                   {0, {SecondMessage_init_zero}}

/* Field tags (for use in manual encoding/decoding) */
#define SecondMessage_test_tag                   1
#define ThirdMessage_test_tag                    1
#define FirstMessage_second_tag                  1
#define FirstMessage_third_tag                   2

/* Struct field encoding specification for nanopb */
extern const pb_field_t ThirdMessage_fields[2];
extern const pb_field_t SecondMessage_fields[2];
extern const pb_field_t FirstMessage_fields[3];

/* Maximum encoded size of messages (where known) */
#define ThirdMessage_size                        2
/* SecondMessage_size depends on runtime parameters */
#define FirstMessage_size                        (0 + (SecondMessage_size > 4 ? SecondMessage_size : 4))

/* Message IDs (where set with "msgid" option) */
#ifdef PB_MSGID

#define SIMPLE_MESSAGES \

#endif

#ifdef __cplusplus
} /* extern "C" */
#endif
/* @@protoc_insertion_point(eof) */

#endif
`

source:
`/* Automatically generated nanopb constant definitions */
/* Generated by nanopb-0.4.0-dev at Tue Oct 16 17:56:36 2018. */

#include "simple.pb.h"

/* @@protoc_insertion_point(includes) */
#if PB_PROTO_HEADER_VERSION != 30
#error Regenerate this file with the current version of nanopb generator.
#endif

const pb_field_t ThirdMessage_fields[2] = {
    PB_FIELD(  1, BOOL    , REQUIRED, STATIC  , FIRST, ThirdMessage, test, test, 0),
    PB_LAST_FIELD
};

const pb_field_t SecondMessage_fields[2] = {
    PB_FIELD(  1, BOOL    , REPEATED, CALLBACK, FIRST, SecondMessage, test, test, 0),
    PB_LAST_FIELD
};

const pb_field_t FirstMessage_fields[3] = {
    PB_ONEOF_FIELD(datails,   1, MESSAGE , ONEOF, STATIC  , FIRST, FirstMessage, second, second, &SecondMessage_fields),
    PB_ONEOF_FIELD(datails,   2, MESSAGE , ONEOF, STATIC  , UNION, FirstMessage, third, third, &ThirdMessage_fields),
    PB_LAST_FIELD
};

/* Check that field information fits in pb_field_t */
#if !defined(PB_FIELD_32BIT)
/* If you get an error here, it means that you need to define PB_FIELD_32BIT
 * compile-time option. You can do that in pb.h or on compiler command line.
 * 
 * The reason you need to do this is that some of your messages contain tag
 * numbers or field sizes that are larger than what can fit in 8 or 16 bit
 * field descriptors.
 */
PB_STATIC_ASSERT((pb_membersize(FirstMessage, datails.second) < 65536 && pb_membersize(FirstMessage, datails.third) < 65536), YOU_MUST_DEFINE_PB_FIELD_32BIT_FOR_MESSAGES_ThirdMessage_SecondMessage_FirstMessage)
#endif

#if !defined(PB_FIELD_16BIT) && !defined(PB_FIELD_32BIT)
/* If you get an error here, it means that you need to define PB_FIELD_16BIT
 * compile-time option. You can do that in pb.h or on compiler command line.
 * 
 * The reason you need to do this is that some of your messages contain tag
 * numbers or field sizes that are larger than what can fit in the default
 * 8 bit descriptors.
 */
PB_STATIC_ASSERT((pb_membersize(FirstMessage, datails.second) < 256 && pb_membersize(FirstMessage, datails.third) < 256), YOU_MUST_DEFINE_PB_FIELD_16BIT_FOR_MESSAGES_ThirdMessage_SecondMessage_FirstMessage)
#endif

/* @@protoc_insertion_point(eof) */
`

PetteriAimonen, 2018-10-16:
Ah yeah, didn't remember, you need to apply it to the whole message like this:

    message FirstMessage {
      option (nanopb_msgopt).no_unions = true;
      oneof datails {
          SecondMessage second = 1;
          ThirdMessage third = 2;
      }
    }

DenisArmy, 2018-10-17:
It works now. Thanks! You save my day!

DenisArmy, 2018-10-19:
Should I close the issue?

closed 2018-10-20 by PetteriAimonen