nanopb/nanopb #354: Decode issue with oneof - \0 prefix added

euripedesrocha, 2018-10-24:
I've been trying to decode the following message:

``` cpp
enum CommandType {
  Comm0 = 0;
  CommNeedsExtra0 = 1;
  CommNeedsExtra0 = 1;
}

message Extra0 {
  string info0 = 1;
  string info1 = 2;
  string info2 = 3;
  string info3 = 4;
}

message Extra1 {
  string info0 = 1;
  string info1 = 2;
  string info2 = 3;
  string info3 = 4;
}

message Command {
  CommandType type = 1;
  oneof parameters {
    Extra0 ext0 = 2;
    Extra1 ext1 = 3;
  }
}
```

- Each field has an entry for it's size in options.
- An earlier version of messages that do not uses oneof works without issues.  
- nanopb version is 0.3.9.1
- compiler is xtensa-esp32-elf-gcc 5.2.0

When decoding the message the decoding process works but each string field has a `\0` prefix that breaks my access to the information. 

As said without the oneof field as in:

``` cpp
message Command {
  CommandType type = 1;
  Extra0 ext0 = 2;
  Extra1 ext1 = 3;
 }
```
Any suggestions on how to fix this?

PetteriAimonen, 2018-10-25:
Can you include a hex dump of the message data you are trying to decode?

PetteriAimonen, 2018-10-25:
(It came to my mind that I've seen a similar problem in the past. It was caused by #define PB_FIELD_16BIT in one .c file but not in the others. All PB_ defines should be made either in pb.h or on the compiler command line or project global settings.)

euripedesrocha, 2018-10-25:
I can produce a hex dump for you. But I have indeed PB_FIELD_16BIT defined, but I have it defined at makefile. I'll check this first.
Thanks.

PetteriAimonen, 2018-11-10:
Do you still have this problem?

closed 2019-01-25 by PetteriAimonen