nanopb/nanopb #355: Proto3 submessage extra data added

crabel99, 2018-10-26:
# Version 
0.4.0-dev

# Description of Problem
Below is the protobuf file that I am using. The problem occurs in the generation of the submessage, an extra byte, `0x04`, gets prepended to each submessage resulting in an `invalid wire_type` error.

# Detailed Explanation and Steps to Reproduce
## probuf file `RCB.proto`
```protobuf
syntax = "proto3";
package RCB;
message ClientMessage {
    uint32 mac = 1;       // This is the last 4 digits of the MAC address
    float InletTemp = 2;  // inlet temperature
    float BoardTemp = 3;  // Embedded board temperature
    bool Flow = 4;        // Reports if adequate flow is available
    bool PowerAvail = 5;  // Reports if control power is available 
    bool IPRFlag = 6;     // flag button
    
    message Status {
        uint32 id = 1;
        bool Power = 2;     // turned on
        bool Fault = 3;     // fault condition reported
        bool Interlock = 4; // installed interlock true
        bool Error = 5;     // true if NOT present
    }

    repeated Status device = 7; // repeat this for each device as needed.
    uint32 time = 8;  // UNIX time of the message
    uint32 address = 9; //  physical address
}
```
## 
For mac = `4c:ba:7c:5f` it is properly encoded as `\0x08\0xdf\0xf8\0xe9\0xe5\0x04`
a sub message, id = 1 and Interlock = 1 should be encoded as `\0x08\0x01\0x20\0x01` it is instead encoded as `\0x04\0x08\0x01\0x20\0x01`. This occurs exactly the same regardless of the number of submessages that are encoded.

This is being developed on an Arduino
```cpp
void mqtt_send() {
  uint8_t sMsg[MSG_SIZE];
  RCB_ClientMessage msg = RCB_ClientMessage_init_zero;
  pb_ostream_t buffer = pb_ostream_from_buffer(sMsg, MSG_SIZE);

  Conv32 bigmac; // simple union to go from the byte array to uint32
  bigmac.bytes[0] = mac[5];
  bigmac.bytes[1] = mac[4];
  bigmac.bytes[2] = mac[3];
  bigmac.bytes[3] = mac[2];
  msg.mac = bigmac.value;                          // field 1 
  msg.device.funcs.encode = mqtt_encode_callback;  // field 7

  //Encode the message
	if (!pb_encode(&buffer, RCB_ClientMessage_fields, &msg)) {
		Serial.println(F("Encoding failed"));
		Serial.println(PB_GET_ERROR(&buffer));
  }

  // publish the buffer
  // ...
 
  //Check to make sure everything decodes correctly
  msg = RCB_ClientMessage_init_default;
  pb_istream_t stream = pb_istream_from_buffer(sMsg, buffer.bytes_written);

  msg.device.funcs.decode = mqtt_decode_callback;
  if (!pb_decode(&stream, RCB_ClientMessage_fields, &msg)) { 
    Serial.print(F("Decode Error: "));
    Serial.println(PB_GET_ERROR(&stream));
    return;
  }
}
```
Here are the callbacks
``` cpp
bool mqtt_decode_callback(pb_istream_t *stream, const pb_field_t *field, void * const **arg) {
  RCB_ClientMessage_MinerStatus msg = RCB_ClientMessage_Status_init_default;
  if (!pb_decode(stream, RCB_ClientMessage_Status_fields, &msg))
    return false;
  return true;
}

bool mqtt_encode_callback(pb_ostream_t *stream, const pb_field_t *field, void * const *arg) {
  RCB_ClientMessage_Status msg;
  for (uint8_t i = 0; i < 2 * shelfNumber; i++){
    msg.id = i + 1;                    // field 1
    msg.Power = power_read(i);         // field 2
    msg.Fault = fault_read(i);         // field 3
    msg.Interlock = interlock_read(i); // field 4
    msg.Error = error_read(i);         // field 5
    if (!pb_encode_submessage(stream, RCB_ClientMessage_Status_fields, &msg)) 
      return false;
  }
  return true;
}
```

PetteriAimonen, 2018-10-26:
I didn't yet look very carefully, but at a glance it seems your `mqtt_encode_callback` is missing a call to `pb_encode_tag_for_field`. Before calling `pb_encode_submessage()` for the field contents, you should call `pb_encode_tag_for_field()` to write the field header.

See e.g. here: https://github.com/nanopb/nanopb/blob/master/examples/network_server/server.c#L43

crabel99, 2018-10-26:
Changed the encoding callback to:
```cpp
bool mqtt_encode_callback(pb_ostream_t *stream, const pb_field_t *field, void * const *arg) {
  RCB_ClientMessage_Status msg;
  for (uint8_t i = 0; i < 2 * shelfNumber; i++){
    msg.id = i + 1;                    // field 1
    msg.Power = power_read(i);         // field 2
    msg.Fault = fault_read(i);         // field 3
    msg.Interlock = interlock_read(i); // field 4
    msg.Error = error_read(i);         // field 5
    if (!pb_encode_tag_for_field(stream, field))
      return false;
    if (!pb_encode_submessage(stream, RCB_ClientMessage_Status_fields, &msg)) 
      return false;
  }
  return true;
}
```

and it works perfectly. Thank you and I apologize for reporting this as a bug.

closed 2018-10-26 by crabel99