nanopb/nanopb #357: Does nanopb support submessages with callback fields in an onof field?

uraj, 2018-10-31:
The issue is about nanopb 3.9.1.

I know nanopb does not support first-level callback fields in oneof, like the following
```
message Msg {
   oneof data {
        string s = 1;
        int32 i = 2;
    }
};
```
Without setting the maxsize of `s`, the generator will report an error

> Callback fields inside of oneof are not supported (field s)

However, the generator will no complain if the message definition is revised to
```
message Wrapper {
    string s = 1;
};

message Msg {
    oneof data {
        Wrapper w = 1;
        int32 i = 2;
    }
};
```

Even though, the decoding of the wrapped string is not actually successful in my test. 

I've looked at nanopb_generator.py. It seems that the script does not recursively check if any nested fields are of the callback type. Is this intended? If not, I can make a patch to the generator and make it complain about nested callback fields.

uraj, 2018-11-01:
Yeah encoding works, thanks for the tag generated for the union. I tried my self for the second case and can confirm that. And I suppose it should also work for the first case? Correct me if I'm wrong. Then the problem is the behavior of the generator is not consistent. It should either accept both cases or reject them both.

PetteriAimonen, 2018-11-01:
Yeah, maybe it should. But I haven't been too interested in changing it, because it seems more like a temporary fix. I've been hoping to find a reasonable way to make callbacks work with oneofs, but haven't had much time to work on it.

uraj, 2018-11-01:
If not changing the code, can we at least make some clarifications in the documentation? I personally spent a lot of time trying to confirm it was not because I did anything wrong. I believe it could be helpful to many other people as well.

PetteriAimonen, 2018-11-01:
Yeah, if you can make a pull request I can merge that.

PR #358 (2018-11-01): Clarify the usage of callback fields inside oneof sections in the docs
Add a few lines to the docs to resolve #357.
closed 2018-11-01 by PetteriAimonen