nanopb/nanopb #359: Splint tests fail on pb_encode.c

docVed, 2018-11-01:
Not really a bug as such, but more for completeness of the test suite.

The tests/splint seems to not run in the travis CI. I assume this is because splint is not installed.
the tests/splint test suite fails with the below error when splint is installed.

```
splint -f splint/splint.rc /home/nanopb/pb_encode.c 2> build/splint/pb_encode.splint
../pb_encode.c: (in function pb_check_proto3_default_value)
../pb_encode.c:235:41: Right operand of && is non-boolean (pb_ssize_t):
                          ((type) & 0x30) == 0x10 && field->size_offset
  The operand of a boolean operator is not a boolean. Use +ptrnegate to allow !
  to be used on pointers. (Use -boolops to inhibit warning)
```

It may also be worth mentioning in the README.md file that there is the option of installing splint
to run the tests/splint suite.

PetteriAimonen, 2018-11-02:
Thanks. Indeed it looks that for some reason, all of my own build machines have lost `splint` at some point in time.

PetteriAimonen, 2018-11-10:
Fix released in 0.3.9.2.

closed 2018-11-10 by PetteriAimonen