nanopb/nanopb #360: Add bazel rules for nanopb generation

silvergasp, 2018-11-06:
**What should the feature do?**

_Option 1_
Use proto_lang_toolchain() to define a toolchain overriding the default protoc library.
- This feature should allow for nanopb to be used with `cc_proto_library` when imported using Bazel.
- The proto_lang_toolchain should override the default library using `--proto_toolchain_for_cc=` compile flag as detailed [here](https://docs.bazel.build/versions/master/command-line-reference.html).

Documentation for this feature can be found [here](https://docs.bazel.build/versions/master/be/protocol-buffer.html#proto_lang_toolchain).

_Option 2_
An alternative which may be more suitable is to make a new skylark rule `cc_nanopb_library`, that behaves identically to `cc_proto_library`, except using nanopb. This rule would use the nanopb generator explicitly, rather than requiring a command line flag everytime you want to compile a nanopb target, overriding the default protoc implementation. Further this would allow for `cc_proto_library` targets to coexist with the `cc_nanopb_targets`.

**In what situation would the feature be useful?**
This is useful in ensuring that the generated libraries in a bazel project are directly in sync with the proto files. i.e. protoc never needs to be manually run to keep the generated files in sync.

_Option 1_
This would be useful for using nanopb exclusively as the generator for a C++/C Project

_Option 2_
This would be useful when building for a heterogeneous system, i.e. a system with low resources (requires nanopb) and a system with enough resources to use standard c++ protobuf libraries.

I personally believe that Option 2 is a better alternative, as this will work well with the heterogeneous systems I work on. I am open to both alternatives.

**NOTE:** I have already done some work on this, but thought I'd open up a discussion. I am happy to help develop this further. Though there a few bugs I could use some help ironing out.

PetteriAimonen, 2018-11-10:
I have no experience in Bazel at all. But there are already some build files for Bazel: https://github.com/nanopb/nanopb/blob/master/BUILD

Is this different somehow?

silvergasp, 2018-11-11:
The BUILD file that you have mentioned if I am not mistaken is for building the runtime, or supporting libraries that the generated code will use. 

Note that the BUILD file uses a `cc_library` target named `nanopb`. A cc_proto_library differs from this, because rather than pointing to existing `*.cc` files it generates `*.cc` files that are dependant on protobuf files.

Perhaps this can be explained by the code below.
```python
# Uses the generated `*.cc/*.h` files from the dependant foo_cc_proto `cc_proto_library`
cc_library(
    name = "lib",
    deps = [":foo_cc_proto"],
)

# Generates `*.proto.cc` / `*.proto.h` files based on the dependant foo_proto `proto_library`
cc_proto_library(
    name = "foo_cc_proto",
    deps = [":foo_proto"],
)

# Describes a library of .proto files
proto_library(
    name = "foo_proto",
    srcs = ["foo.proto"],
)
````
Using bazel you can override the default usage of protoc to use a plugin (i.e. nanopb), this is done with the `--proto_toolchain_for_cc=`. This is discussed in Option 1 above.

Option 2 would be to make another rule, using skylark or genrule (plugin type languages/tools for bazel), to make another rule that handles nanopb as an entirely separate function. An example of how this might look when used in a BUILD file is shown below.

```python
# Uses the generated `*.cc/*.h` files from the dependant foo_cc_proto `cc_nanopb_proto_library`
cc_library(
    name = "lib",
    deps = [":foo_cc_proto"],
)

# Generates `*.proto.c` / `*.proto.h` files based on the dependant foo_proto `proto_library`
cc_nanopb_proto_library( #NOTE: this rule name has changed to handle nanopb generation 
    name = "foo_cc_proto",
    deps = [":foo_proto"],
)

# Describes a library of .proto files
proto_library(
    name = "foo_proto",
    srcs = ["foo.proto"],
)
````
The motivation behind doing this is that your nanopb generated `*.c\*.h` files are kept in sync with your .proto files. This is useful because it avoids a situation where you might think that your generated files are up to date with your `.proto` files, when they are in fact not.

Once I have ironed out the kinks in my current implementation of `cc_nanopb_proto_library`, I will make a pull request and go from there. At the moment I am more just testing the waters to see if there is interest, and to see if people had any input/better ideas.

llhe, 2018-12-26:
@silvergasp This is great feature to have.

kalbasit, 2019-05-10:
@silvergasp any progress on getting `cc_nanopb_proto_library` implemented?

silvergasp, 2019-05-11:
@kalbasit At the moment there is some significant restructuring going on to provide a starlark API for c++ rules in bazel (see [here](https://github.com/bazelbuild/bazel/issues/4570)). There are a couple of potential ways I can see that this could be done.
1. Use the yet to be WIP starlark api for c++ rules in bazel. [These rules](https://github.com/bazelbuild/rules_cc/blob/master/cc/rules.bzl) currently provide a shim of the the native bazel rules, which will be re-implemented in starlark. Once this is done I would then re-configure the starlark version of [cc_proto_library](https://github.com/bazelbuild/rules_cc/blob/b308aae57fe187c670e465591081177dbfb95960/cc/rules.bzl#L57) to work with nanopb. I believe this to be the simplest and best solution long term. However will require waiting ~6-12months or so for this part of bazel to be restructured.

2. Use protobufs [proto_gen rule](https://github.com/protocolbuffers/protobuf/blob/982d5d2ab1e41a0071b9514cad55cca00149cd5c/protobuf.bzl#L176). This would be a little more of a hack job and should be deprecated when option 1^ is implemented, this will introduce some breaking changes down the line. The advantage of this is that it can be done now. This is close to what I have got setup at the moment however this will only work on my particular setup on my local machine. Therefore I would need to package it up into a collection of portable rules. Not far off at the moment, however have had to place priorities on other projects for the moment.

At the moment I am waiting for bazels starlark API's to stabilise so that I can provide a stable API.

I would be happy to share the approach I have taken to date, however I don't think creating a pull request at this time would be appropriate without being sure that the API's I am using are stable.

kalbasit, 2019-05-11:
@silvergasp thank you for the detailed response, very helpful! My timeline is short-term so I'm probably going to go with option 2. It would be really great if you could provide a simple example using it for generating nanopb protobuf that would be great. I know it's hacky, but I'm willing to cut any shortcut to get it to work now and fix it later.

Much appreciated!

ArjunNarayan2066, 2019-06-23:
Hi, I was looking into implementing this into my own work. I began by building my own Bazel target (`def c_proto_library`) consisting of a genrule call & cc_library to package. However this fizzled out after I realized the complexity required to build from scratch to address the naming conflict (protoc C++ & nanopb both make header files called person.pb.h). I then tried implementing my own target that calls the `proto_gen_rule` rule as described above. However I ran into the issue where the `plugin` field requires a Bazel label to run. This means I need to wrap the `protoc-gen-nanopb` shell script in a `sh_binary` target, but whenever I do this is gives errors saying: `Label '//dir_to/third_party:nanopb/generator/protoc-gen-nanopb' crosses boundary of subpackage 'dir_to/third_party/nanopb'` and others too. Has anyone had any better luck addressing these issues?

kalbasit, 2019-08-08:
@ArjunNarayan2066 Can you share your code so far? Maybe we can collaborate on it.

silvergasp, 2020-03-17:
@kalbasit @ArjunNarayan2066, I have a basic proof of concept working for this at the moment, and have just opened a pull request. Feel free to use my fork in the mean time, or contribute to improvements in this feature.

PetteriAimonen, 2022-11-25:
The bazel rules were finally merged a few months ago, so I'm closing off this issue.
Thanks for everyone and especially silvergasp for patience!

closed 2022-11-25 by PetteriAimonen