nanopb/nanopb #361: Getting segmentation fault while encryption when accessing through shared library (.so)  in ARM

ganeshthorner, 2018-11-06:
dlInit called..
Create Broker Instance and Connect..client1:myganesh1
LOG: Client myganesh1 sending CONNECT
[New Thread 1106846832 (LWP 2258)]
LOG: Client myganesh1 received CONNACK (0)
Broker Connect Success 162.255.87.22:1883
dlScanFirst Called..
Current Sequence Number: 0
Adding metric: 'Node Control/Next Server'
Setting datatype and value - value size is 1
Setting datatype: 11, with value: 0
Size of metrics payload 1
Adding metric: 'Node Control/Rebirth'
Setting datatype and value - value size is 1
Setting datatype: 11, with value: 0
Size of metrics payload 2
Adding metric: 'Node Control/Reboot'
Setting datatype and value - value size is 1
Setting datatype: 11, with value: 0
Size of metrics payload 3
Adding metric: 'Node Metric0'
Setting datatype and value - value size is 11
Setting datatype: 12, with value: hello node
Size of metrics payload 4
Adding metric: 'Node Metric1'
Setting datatype and value - value size is 1
Setting datatype: 11, with value: 1
Size of metrics payload 5
column_keys[i]: Int8s
column_keys[i]: Int16s
column_keys[i]: Int32s
Adding metric: 'DataSet'
Setting datatype and value - value size is 48
Setting datatype: 16, with value: 1
Adding metric: 'Node Metric2'
Setting datatype and value - value size is 4
Setting datatype: 2, with value: 13
Setting datatype: 12, with value: MyCustomUnits
Size of values in PropertySet 1
Setting datatype and value - value size is 4
Setting datatype: 3, with value: 0
Setting datatype and value - value size is 4
Setting datatype: 3, with value: 0
Setting datatype and value - value size is 32
Setting datatype: 19, with # of metrics: 2
Payload:  has_timestamp: true
Payload:  timestamp: 1541606935193
Payload:  has_seq: true
Payload:  seq: 0
Payload:  UUID: MyUUID
Payload:  Size of metric array: 8
Payload:  Metric 0:  name: Node Control/Next Server
Payload:  Metric 0:  has_alias: true
Payload:  Metric 0:  alias: 0
Payload:  Metric 0:  has_timestamp: true
Payload:  Metric 0:  timestamp: 1541606935194
Payload:  Metric 0:  has_datatype: true
Payload:  Metric 0:  datatype: 11
Payload:  Metric 0:  has_is_historical: false
Payload:  Metric 0:  has_is_transient: false
Payload:  Metric 0:  has_is_null: false
Payload:  Metric 0:  has_metadata: false
Payload:  Metric 0:  has_properties: false
Payload:  Metric 0:  datatype: 11, with value: 0
Payload:  Metric 1:  name: Node Control/Rebirth
Payload:  Metric 1:  has_alias: true
Payload:  Metric 1:  alias: 1
Payload:  Metric 1:  has_timestamp: true
Payload:  Metric 1:  timestamp: 1541606935194
Payload:  Metric 1:  has_datatype: true
Payload:  Metric 1:  datatype: 11
Payload:  Metric 1:  has_is_historical: false
Payload:  Metric 1:  has_is_transient: false
Payload:  Metric 1:  has_is_null: false
Payload:  Metric 1:  has_metadata: false
Payload:  Metric 1:  has_properties: false
Payload:  Metric 1:  datatype: 11, with value: 0
Payload:  Metric 2:  name: Node Control/Reboot
Payload:  Metric 2:  has_alias: true
Payload:  Metric 2:  alias: 2
Payload:  Metric 2:  has_timestamp: true
Payload:  Metric 2:  timestamp: 1541606935194
Payload:  Metric 2:  has_datatype: true
Payload:  Metric 2:  datatype: 11
Payload:  Metric 2:  has_is_historical: false
Payload:  Metric 2:  has_is_transient: false
Payload:  Metric 2:  has_is_null: false
Payload:  Metric 2:  has_metadata: false
Payload:  Metric 2:  has_properties: false
Payload:  Metric 2:  datatype: 11, with value: 0
Payload:  Metric 3:  name: Node Metric0
Payload:  Metric 3:  has_alias: true
Payload:  Metric 3:  alias: 4
Payload:  Metric 3:  has_timestamp: true
Payload:  Metric 3:  timestamp: 1541606935195
Payload:  Metric 3:  has_datatype: true
Payload:  Metric 3:  datatype: 12
Payload:  Metric 3:  has_is_historical: false
Payload:  Metric 3:  has_is_transient: false
Payload:  Metric 3:  has_is_null: false
Payload:  Metric 3:  has_metadata: false
Payload:  Metric 3:  has_properties: false
Payload:  Metric 3:  datatype: 12, with value: hello node
Payload:  Metric 4:  name: Node Metric1
Payload:  Metric 4:  has_alias: true
Payload:  Metric 4:  alias: 5
Payload:  Metric 4:  has_timestamp: true
Payload:  Metric 4:  timestamp: 1541606935195
Payload:  Metric 4:  has_datatype: true
Payload:  Metric 4:  datatype: 11
Payload:  Metric 4:  has_is_historical: false
Payload:  Metric 4:  has_is_transient: false
Payload:  Metric 4:  has_is_null: false
Payload:  Metric 4:  has_metadata: false
Payload:  Metric 4:  has_properties: false
Payload:  Metric 4:  datatype: 11, with value: 1
Payload:  Metric 5:  name: DataSet
Payload:  Metric 5:  has_alias: true
Payload:  Metric 5:  alias: 3
Payload:  Metric 5:  has_timestamp: false
Payload:  Metric 5:  has_datatype: true
Payload:  Metric 5:  datatype: 16
Payload:  Metric 5:  has_is_historical: false
Payload:  Metric 5:  has_is_transient: false
Payload:  Metric 5:  has_is_null: false
Payload:  Metric 5:  has_metadata: false
Payload:  Metric 5:  has_properties: false
Payload:  Metric 5:  datatype DATASET - Not yet supported
Payload:  Metric 6:  name: Node Metric2
Payload:  Metric 6:  has_alias: true
Payload:  Metric 6:  alias: 6
Payload:  Metric 6:  has_timestamp: false
Payload:  Metric 6:  has_datatype: true
Payload:  Metric 6:  datatype: 2
Payload:  Metric 6:  has_is_historical: false
Payload:  Metric 6:  has_is_transient: false
Payload:  Metric 6:  has_is_null: false
Payload:  Metric 6:  has_metadata: false
Payload:  Metric 6:  has_properties: true
Payload:  Metric 6:  datatype: 2, with value: 13
Payload:  Metric 7:  name: _types_/Custom_Motor
Payload:  Metric 7:  has_alias: false
Payload:  Metric 7:  has_timestamp: false
Payload:  Metric 7:  has_datatype: true
Payload:  Metric 7:  datatype: 19
Payload:  Metric 7:  has_is_historical: false
Payload:  Metric 7:  has_is_transient: false
Payload:  Metric 7:  has_is_null: false
Payload:  Metric 7:  has_metadata: false
Payload:  Metric 7:  has_properties: false
Payload:  Metric 7:  datatype: 19, with # of metrics: 2
Encoding...

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 1073872608 (LWP 2225)]
pb_field_iter_begin (iter=0xbef34410, fields=0x4130, dest_struct=0x2f12368)
    at pb_common.c:14
14      pb_common.c: No such file or directory.
        in pb_common.c
(gdb) bt
#0  pb_field_iter_begin (iter=0xbef34410, fields=0x4130, dest_struct=0x2f12368)
    at pb_common.c:14
#1  0x412f9224 in pb_encode (stream=0xbef3443c, fields=0x4130,
    src_struct=0x2f12368) at pb_encode.c:395
#2  0x412f9330 in pb_encode_submessage (stream=0xbef344dc, fields=0x4130,
    src_struct=0x2f12368) at pb_encode.c:548
#3  0x412f8c20 in encode_field (stream=0xbef344dc, field=0x41306761,
    pData=0x2f12368) at pb_encode.c:190
#4  0x412f92cc in pb_encode (stream=0xbef344dc, fields=<value optimized out>,
    src_struct=<value optimized out>) at pb_encode.c:408
#5  0x412fa768 in encode_payload (buffer=<value optimized out>,
    buffer_length=<value optimized out>, payload=0xbef34918)
    at sparkplug_b.c:540
#6  0x412f6920 in frameNodeBirth (spPtr=<value optimized out>) at dpcLib.c:274

PetteriAimonen, 2018-11-06:
Hmm, the cause of crash is that fields pointer value 0x4130 is invalid.

It becomes invalid between steps 2 and 3 in the backtrace, most likely when `pb_enc_submessage()` takes the submessage fields pointer from the main message pb_field_t array.

Because the correct fields value looks to be `0x4130????`, it seems to be shifted right by 16 bits. This suggests that the `pb_field_t` definition used by `pb_encode.c` is different than the one used by `.pb.c` file.

Are you defining `PB_FIELD_16BIT` or `PB_NO_PACKED_STRUCTS` somewhere? Make sure that any such configuration options are project-global, i.e. done either on compiler command line or by editing pb.h.

ganeshthorner, 2018-11-07:
Thank you .. it has been fixed by enabling PB_NO_PACKED_STRUCTS in pb.h

closed 2018-11-07 by ganeshthorner