nanopb/nanopb #362: proto_path option does not work on Windows

stsymbaliuk, 2018-11-13:
When call nanopb from root of project with `--proto_path` it can't find proto file in subfolder
```
e:\nanopb_project>e:\nanopb-0.3.9.2-windows-x86\generator-bin\protoc.exe --proto_path=proto_files --cpp_out=proto_out_c simple.proto
simple.proto: No such file or directory

e:\nanopb_project>e:\nanopb-0.3.9.2-windows-x86\generator-bin\protoc.exe --version
libprotoc 3.1.0
```

But Google protobuf works fine with the same parameters:
```
e:\nanopb_project>e:\protoc-3.6.1\bin\protoc.exe --proto_path=proto_files --cpp_out=proto_out_c simple.proto
```

Google protobuf generate files in `proto_out_c` folder 
```
e:\nanopb_project>tree /F /A
+---proto_files
|       simple.proto
|
\---proto_out_c
        simple.pb.cc
        simple.pb.h
		
e:\nanopb_project>e:\protoc-3.6.1\bin\protoc.exe --version
libprotoc 3.6.1
```

To make work nanopb I can call it from `proto_files` folder
```
e:\nanopb_project\proto_files>e:\nanopb-0.3.9.2-windows-x86\generator-bin\protoc.exe --proto_path=. --cpp_out=../proto_out_c simple.proto
```
When I call nanopb protoc.exe and google one from proto_files with `--proto_path=../proto_files` I get next results:

Error with nanopb proto.c:
```
e:\nanopb_project\proto_files>e:\nanopb-0.3.9.2-windows-x86\generator-bin\protoc.exe --proto_path=../proto_files --cpp_out=../proto_out_c simple.proto
simple.proto: File does not reside within any path specified using 
--proto_path (or -I).  You must specify a --proto_path which encompasses this 
file.  Note that the proto_path must be an exact prefix of the .proto file 
names -- protoc is too dumb to figure out when two paths (e.g. absolute and 
relative) are equivalent (it's harder than you think).
```

Works as expected google proto.c:
```
e:\nanopb_project\proto_files>e:\protoc-3.6.1\bin\protoc.exe --proto_path=../proto_files --cpp_out=../proto_out_c simple.proto
```

Same results I get when I set proto_path with absolute path.

I'm using Windows 10 x64

PetteriAimonen, 2018-11-13:
This could be due to the version difference of protoc.exe.

Could you try to copy the newer `protoc.exe` to `generator-bin` folder?

I've recently upgraded the protoc version that gets included in the Windows binary distribution, but 0.3.9.2 still has the old version. Here is a newer one that you can try also, though copying the file should do the same. http://lakka.kapsi.fi:50140/job/nanopb-binary-windows/lastSuccessfulBuild/artifact/dist/nanopb-0.3.9.2-29-g9710d46-windows-x86.zip

stsymbaliuk, 2018-11-13:
> Could you try to copy the newer `protoc.exe` to `generator-bin` folder

Yes it works as expected when I use `protoc.exe` from `nanopb-0.3.9.2-29-g9710d46-windows-x86` folder

I used it with next command:
```
e:\nanopb-0.3.9.2-29-g9710d46-windows-x86\generator-bin\protoc.exe --proto_path=proto_files --nanopb_out=proto_out_c simple.proto
```

PetteriAimonen, 2019-03-07:
Note to self: new protoc.exe version will be included in 0.3.9.3.

PetteriAimonen, 2019-03-08:
Fix released in nanopb-0.3.9.3.

closed 2019-03-08 by PetteriAimonen