nanopb/nanopb #363: Incorrect PB_STATIC_ASSERT for messages containing a "oneof"

lazorenkoigor96, 2018-11-19:
My `.proto` file describing the message looks like this:

```
syntax = "proto2";

import "nanopb.proto";

message BodyMessage {
    oneof body_type {
        bytes device_data_crypted = 1 [(nanopb).max_size = 256];
        bytes device_config_crypted = 2 [(nanopb).max_size = 256];
    }
}
```

Сompiling `.proto`, I got this structure:

```
typedef struct _BodyMessage {
    pb_size_t which_body_type;
    union {
        BodyMessage_device_data_crypted_t device_data_crypted;
        BodyMessage_device_config_crypted_t device_config_crypted;
    } body_type;
/* @@protoc_insertion_point(struct:BodyMessage) */
} BodyMessage;
```
There is a STATIC ASSERT in `.с` file:

```
/* Check that field information fits in pb_field_t */
#if !defined(PB_FIELD_32BIT)
/* If you get an error here, it means that you need to define PB_FIELD_32BIT
 * compile-time option. You can do that in pb.h or on compiler command line.
 * 
 * The reason you need to do this is that some of your messages contain tag
 * numbers or field sizes that are larger than what can fit in 8 or 16 bit
 * field descriptors.
 */
PB_STATIC_ASSERT((pb_membersize(BodyMessage, device_data_crypted) < 65536 && pb_membersize(BodyMessage, device_config_crypted) < 65536), YOU_MUST_DEFINE_PB_FIELD_32BIT_FOR_MESSAGES_BodyMessage)
#endif
```
This ASSERT uses such a macro from a `pb.h`:

```
/* Size of a structure member, in bytes. */
#define pb_membersize(st, m) (sizeof ((st*)0)->m)
```
When I try to build a project, I get errors:
```
'BodyMessage {aka struct _BodyMessage}' has no member named 'device_config_crypted'
'BodyMessage {aka struct _BodyMessage}' has no member named 'device_data_crypted'
```
I think the problem is that ASSERT is generated incorrectly. Since the elements need to be accessed through union. Like this:
```
PB_STATIC_ASSERT((pb_membersize(BodyMessage, body_type.device_data_crypted) < 65536 && pb_membersize(BodyMessage, body_type.device_config_crypted) < 65536), YOU_MUST_DEFINE_PB_FIELD_32BIT_FOR_MESSAGES_BodyMessage)
```

PetteriAimonen, 2018-11-19:
Yeah, looks like that's what is happening. Bug is here:
https://github.com/nanopb/nanopb/blob/maintenance_0.3/generator/nanopb_generator.py#L620
It has separate path for doing it correctly for submessages, but the same thing with member name should be done for any field type.

It looks like this logic will be changed anyway for 0.4.0 release, but it might be worth it to fix in the maintenance branch. For a temporary workaround, you can also define PB_NO_STATIC_ASSERT to disable all the checks.

PetteriAimonen, 2019-03-07:
Fixed in maintenance_0.3 now, fix will be included in 0.3.9.3.

PetteriAimonen, 2019-03-08:
Fix released in nanopb-0.3.9.3.

closed 2019-03-08 by PetteriAimonen