nanopb/nanopb #364: Encode/Decode sub messages without callbacks

masoudr, 2018-11-19:
Hi,
I'm new to `nanopb`. I'm trying to create a simple example with nested messages but It seems that encoding is fine but when I want to decode it I get `end-of-stream` error. If I understand correctly I only need callbacks if I define reapeated messages or undefined string size message. Is my encoding steps is correct? How about decoding which one should I decode first submessages (child) or main message?
Thanks.
Suppose this example:
```
# my_sample.proto
message MainMessage {
    message Child {
        required int32 a = 1;
        required in32 b = 2;
    }
    required Child child = 1;
}

```

```
void decode(uint8_t *input, size_t message_length)
{
    MainMessage msg = MainMessage_init_zero;
    pb_istream_t stream = pb_istream_from_buffer(input, message_length);
    bool status = pb_decode(&stream, MainMessage_Child_fields, &msg.child);
    if (!status)
    {
        printf("Decoding failed: %s\n", PB_GET_ERROR(&stream));
        return;
    }
    status = pb_decode(&stream, MainMessage_fields, &msg);
    if (!status)
    {
        printf("Decoding failed: %s\n", PB_GET_ERROR(&stream));
        return;
    }
    printf("Message decoded: %s\n", msg.child.a);
}

void encode()
{
    MainMessage msg = MainMessage_init_zero;
    uint8_t buffer[128] = {0};
    size_t message_length = 0;
    bool status;
    pb_ostream_t stream = pb_ostream_from_buffer(buffer, sizeof(buffer));

    if (! pb_encode_submessage(&stream, MainMessage_Child_fields, &msg.child)){
        printf("pb_encode_submessage Failed!\n");
        return;
    }

    if (!(status = pb_encode(&stream, MainMessagea_fields, &msg)))
    {
        printf("pb_encode Encoding failed: %s\n", PB_GET_ERROR(&stream));
        return;
    }
    message_length = stream.bytes_written;
    printf("buffer: ");
    for (int i = 0; i < sizeof(buffer); i++)
    {
        if (buffer[i])
            printf("%d, ", buffer[i]);
    }
    printf("\nmessage_length: %d\n", message_length);
    decode(buffer, message_length);
}

int main()
{
    encode();
}
```
Thanks.

PetteriAimonen, 2018-11-19:
You don't need to separately encode/decode the submessage.

Just the parts that encode the mainmessage is enough, the submessage will automatically be encoded as part of it.

masoudr, 2018-11-19:
@PetteriAimonen I already tried that but after decoding the values for sub-messages is all default ones. I edited the main code I get zeroes for `a` and `b`.

PetteriAimonen, 2018-11-19:
For optionals, you need to set has_child = true.

masoudr, 2018-11-20:
@PetteriAimonen Thanks, but where should I add this param? Can you give me an example?

uraj, 2018-11-20:
@masoudr Check out [this](https://jpa.kapsi.fi/nanopb/docs/reference.html#proto-file-options).

masoudr, 2018-11-20:
@uraj I still can't get it. Do you mind adding it to my current example?

uraj, 2018-11-20:
@masoudr Sorry my previous comment was wrong. I've edited it. Check it out.

masoudr, 2018-11-20:
@uraj Thanks. Do I need to set it for encoding too?

uraj, 2018-11-20:
You set it only for encoding. When decoding, the decoder sets it for you so you know whether the incoming message contains the optional or not.

closed 2018-11-20 by masoudr

masoudr, 2018-11-22:
Can you tell me what option should I set when I use a `repeated` field? It seems that there is no option with `has_...` for the repeated fields.

PetteriAimonen, 2018-11-22:
Don't be afraid to look inside the generated .pb.h files. You'll find a "field_count" field there.