nanopb/nanopb #367: Options file: Differently interpreted on Posix / Windows

ceggers-arri, 2018-12-01:
message.proto:
```
syntax = "proto2";

message Outer
{
	message Inner
	{
		required uint32 member1 = 1;
		required uint32 member2 = 2;
	}  // message Inner

	optional Inner inner = 1;

}  // message Outer
```
message.options:
`Outer.inner       type:FT_CALLBACK`

message.pb.h (Posix, considered correct):
```
...
/* Struct definitions */
typedef struct _Outer {
    pb_callback_t inner;
/* @@protoc_insertion_point(struct:Outer) */
} Outer;

typedef struct _Outer_Inner {
    uint32_t member1;
    uint32_t member2;
/* @@protoc_insertion_point(struct:Outer_Inner) */
} Outer_Inner;
...
```

mesage.pb.h (Windows, considered wrong):
```
...
/* Struct definitions */
typedef struct _Outer {
    pb_callback_t inner;
/* @@protoc_insertion_point(struct:Outer) */
} Outer;

typedef struct _Outer_Inner {
    pb_callback_t member1;
    pb_callback_t member2;
/* @@protoc_insertion_point(struct:Outer_Inner) */
} Outer_Inner;
...
```

On both systems, `Outer.inner` is generated as callback (as specified in the options file). Only on Windows, additionally `Outer_Inner.member1` and `Outer_Inner.member2` are generated als callback, which has not been specified.

I guess that fnmatch( ) behaves case sensitive on Posix and case insensitive on Windows, which would explain the difference.

Temporary solution: Using Posix (msys) version of python(2) on Windows.
Intended solution: Identical generated code on both platforms.
Tested on: Windows (with msys2 and msys2/mingw64), Linux

PetteriAimonen, 2018-12-01:
Yeah, I noticed same problem a few weeks ago, it is fixed here:
https://github.com/nanopb/nanopb/commit/c8274d6a6571abdca91485f07f422fb5767afef4

ceggers-arri, 2018-12-03:
Works fine for me :+1: . Thank you very much for quick response on the weekend!

closed 2018-12-03 by ceggers-arri