nanopb/nanopb #368: Proper way to send an encoded data

masoudr, 2018-12-12:
Hi,
I'm trying to send an encoded buffer (array of `unsigned char`) from a server to a client with an http request. My first attempt was to convert the array of `unsigned char` to equivalent ASCII string and then send throw the network. for example, suppose this buffer:

    unsigned char buffer[] = {72, 101, 108, 108, 111, 32, 119, 111, 114, 108, 100, 0, 24, 1, 32, 1, 40, 0};
And if I convert it to a string I would have this:

> Hello world  (

which has null terminating chars and I have problems on the client side to read it back because I can't find the end of the message and reconstruct the buffer. Is there any way to create a string from ProtoBuf output with minimal length?

PetteriAimonen, 2018-12-12:
I see you've already asked the same question on StackOverflow ( https://stackoverflow.com/questions/53736331/input-string-consists-of-null-terminating-chars ).

Please wait to see if the answers you get there help you forwards.

Even `SerializeToString` in C++ protobuf library will have null bytes in the string. C++ strings can store null bytes, while C strings cannot. Encoded protobuf data is always binary data and you have to treat it as such, i.e. don't use string functions on it.

closed 2018-12-12 by PetteriAimonen

masoudr, 2018-12-12:
@PetteriAimonen OK, but what is the proper way to send them over a network?

PetteriAimonen, 2018-12-12:
https://github.com/nanopb/nanopb/tree/master/examples/network_server  For example like this example does it.

masoudr, 2018-12-12:
@PetteriAimonen I saw that example but can't understand how you send an `unsigned char` array, can you explain a little bit?
@KidVizious Thank you it is a good idea but I don't have enough resource to encode to base64.

KidVizious, 2018-12-12:
@masoudr Obviously, I don't know your resource requirements, but you can Base64 encode with about 50 lines of code.

masoudr, 2018-12-12:
@KidVizious Thanks, sure I will give it a shot.