nanopb/nanopb #371: Add #define for defaults that works with assignment to pointer

rubypanther, 2018-12-29:
**What should the feature do?**
Currently, in the generated files only one `#define` is added for assigning defaults to the struct.
Something like:
`#define MyMessage_init_default               {"", ""}`
Which can be used as in:
`MyMessage foo = MyMessage_init_default;`

It would be nice to have another one for assigning via pointers:
`#define MyMessage_init_default_ptr               (MyMessage){.fieldone="",.fieldtwo=""}`
Which could then be used like:
`MyMessage *foo = malloc(sizeof(MyMessage));`
`*foo = MyMessage_init_default_ptr;`

**In what situation would the feature be useful?**
right now it is hard to get at the default values if you're allocating on the heap. When allocating to the stack, if you have large repeated fields with `max_size` and `max_count` set it is easy to segfault.

Right now I'm leaving off the `max_count` and using the callback interface for that field, but that's a rather heavy workaround. :)

`MyMessage_init_zero` probably isn't as needed because you can just use `calloc` or equivalent, though somebody might expect it to be there too.

PetteriAimonen, 2018-12-30:
Hmm, doesn't `*foo = (MyMessage)MyMessage_init_default` work the same as the proposed format?

Designated initializers are a C99-only feature, so I generally try to avoid them in nanopb code. For example, some users compile nanopb code using old embedded compilers, or e.g. C++11 compilers which do not support them.

Though do note that `pb_decode()` does apply default values itself also, so often just using `calloc()` would be enough.

rubypanther, 2018-12-31:
It's a syntax error, unfortunately.

The C99 way is the only way to get it one line, AFAIK. I guess there are a bunch of ways to achieve the same thing with a function instead of a macro, and that could be more portable. But probably more work. And also, if you're on an old compiler you're probably less likely to want to use the heap, since that usually implies a really small micro.

My goal is to have an internal library that runs on both larger and smaller systems, where the smaller system might only be decoding the required, non-repeating fields. But I'd still like to be able to just dump the repeating fields into memory on the larger systems, especially during development. Then later I can write callbacks to make it more efficient, or to fit it into the smaller systems.

It would only be useful as an option, but probably gcc use is common.

I can look at the pb_decode() source to see how it is done there. For decode it doesn't really matter, but for encode it is sometimes nice to be able to check what data will get sent out before sending it out.

rubypanther, 2019-01-11:
You're right, thanks! I was getting `error: expected expression before ‘{’ token` but the cast fixes it. The answers I was finding online all said I had to use the C99 technique to avoid looping over the members. Sorry for the useless request. :)

closed 2019-01-11 by rubypanther