nanopb/nanopb #372: Unnecessary warning

alexsho, 2019-01-01:
In my .proto file, i have two enums with same field name.
When translating the enums from .proto to C enum, the enum's name is added to each field.
Yet i get warning/error message that some field should be unique within the global scope.
I know that it is not a good practice to define same name, but once it is translated to C as unique name, it should not be forbidden.

PetteriAimonen, 2019-01-01:
Could you include the actual, complete error message you are getting?

alxsfrm, 2019-01-02:
Sorry, for that.
Here is the part of the proto:
`enum Channels`
`{`
  `    ...`
`  Rth = 4;`
  `    ...`
`} `

And here is the actual error message:
`my_icd.proto:350:5: "Rth" is already defined.`
`my_icd.proto:350:5: Note that enum values use C++ scoping rules, meaning that enum values are siblings of their type, not children of it. therefore, "Rth" must be unique within the global scope, not just within "Channels".`
When i write my C code, usually i put the enum name as part of the field name, exactly like the protoc compiler does, but if i'll do the same in order to comply with the error, i'll see in generated C file the enum name appears twice, like `Channels_Channels_Rth`.

PetteriAimonen, 2019-01-02:
Yeah, that message is from `protoc`. It makes sense as the .proto file should be portable across different languages and nanopb cannot influence that part of protoc behaviour anyway.

You can disable the enum name prefix using `long_names:false` option for nanopb generator. E.g. `* long_names:false` in `.options` file to set it for all fields.

It would have made sense for nanopb to use the same names as C++ does to start with, but back then I didn't realize the difference and I haven't changed it to keep compatibility. Perhaps sometime..

closed 2019-01-25 by PetteriAimonen