nanopb/nanopb #373: Overriding int_size does not affect maximum size of messages

MarioRose, 2019-01-02:
Hi,

as discussed in https://groups.google.com/forum/#!topic/nanopb/ZcqcUaY5UVo

PetteriAimonen, 2019-03-07:
Note to self: the fix doesn't backport easily to 0.3.9.x and it is kind of low priority, so I'll leave this for 0.4.0.

unsignedint, 2019-03-09:
This is actually a problem that we have hit on a project I am are working on (as we now start to optimise memory usage), I can confirm the fix in git works... any idea on when we could expect 0.4.0?

PetteriAimonen, 2019-03-09:
In a few months perhaps. There are some infrastructure problems I need to fix first before I have enough confidence in it, because the AVR & ARM test envs haven't been working lately.

shreyasbharath, 2019-03-22:
Looking forward to this fix!

PetteriAimonen, 2019-03-22:
Well, you can start using the master branch already. Binary packages are available here: http://jpa.kapsi.fi/jenkins/job/nanopb-binary-windows/

closed 2019-12-20 by PetteriAimonen