nanopb/nanopb #374: Message decoding proto3

alxsfrm, 2019-01-09:
Hi, 
In proto3 the default is that all the data fields are optional.
How can i know that some field does exists, in order to read and use it ?
For example in proto2 there was **has_*** boolean field generated, so it was easy to understand. What is the flag in proto3 ?

PetteriAimonen, 2019-01-09:
In proto3, field is present always when its value differs from "zero". So there is no need to check, you can just use the value directly. Basically there is no concept of field not existing in proto3.

https://developers.google.com/protocol-buffers/docs/proto3#default

alxsfrm, 2019-01-09:
Understood.
The problem is that sometimes the value can really be zero.
So how should i know when the sent value was zero, or the sender didn't meant to send it ?

PetteriAimonen, 2019-01-09:
I agree that the proto3 way to treat default values is not always very useful. There are various methods such as oneof and wrappers that will also work with nanopb: https://stackoverflow.com/questions/42622015/how-to-define-an-optional-field-in-protobuf-3

For messages, nanopb implementation in 0.3.x is that message is "not set" when all of its fields have the default value. In 0.4.x this will probably change to have a separate has_ field because that seems more common among other libraries (issue #308).

closed 2019-03-07 by PetteriAimonen