nanopb/nanopb #375: Encoding nested bytes with pb_callback

samdenty, 2019-01-14:
Hi there, how can I encode nested bytes into a message? I see that the bytes field is compiled to a `pb_callback_t`, but I'm unsure on how to use it. I've tried searching through the issues - closest I could find was #163.

**message.proto:**

```proto
message TestEvent {
  int32 test = 1;
}

message MessageEvent {
  bytes data = 1;
}
```

```C++
// Create TestEvent and serialize it to a stream
uint8_t testEventBuffer[128];
TestEvent testEvent = TestEvent_init_zero;
testEvent.test = 100;
pb_ostream_t testEventStream = pb_ostream_from_buffer(testEventBuffer, sizeof(testEventBuffer));
pb_encode(&testEventStream, TestEvent_fields, &testEvent);

// Create a MessageEvent and the testEvent as a field
uint8_t buffer[128];
MessageEvent messageEvent = MessageEvent_init_zero;
pb_ostream_t stream = pb_ostream_from_buffer(buffer, sizeof(buffer));

// How can I assign messageEvent.data with the testEvent bytes
// ie. messageEvent.data = testEvent;
messageEvent.data.funcs.encode(&testEventStream, &testEvent); // Compile error
```

samdenty, 2019-01-14:
@PetteriAimonen Just simplification for the above code, I'm using it as an any type.

PetteriAimonen, 2019-01-14:
Ok, well it is just like encoding any bytes field. You can define `max_size` in options to get a static array in the struct, and then you can give that to `pb_ostream_from_buffer()`. Or you can write out the bytes data in the callback.

Maybe some of these will help:
https://github.com/nanopb/nanopb/blob/master/examples/network_server/server.c#L26
https://groups.google.com/forum/?fromgroups#!topic/nanopb/Nz07zw9SMn4
https://github.com/nanopb/nanopb/blob/master/tests/callbacks/encode_callbacks.c

Note that you don't have to encode to a buffer beforehand, it will take less memory if you do it like in network_server example and call pb_encode_submessage() from inside the callback.